RT-Thread/rt-thread #9891: [Feature] AArch64 full dm 2.0 support

GuEe-GUI, 2025-01-09:
### Describe problem solved by the proposed feature

## 状况

后续计划将在 aarch64 平台强制使用 DM，以改善目前代码混乱，以及 rtos 和 smart 驱动维护不统一的问题，目前在所有 aarch64 平台中，对 DM 的支持情况（包括尚未提交的内容）

- cvitek
- phytium
- qemu-virt64-aarch64（支持）
- raspberry-pi（支持）
- rockchip（支持）
- zynqmp-a53-dfzu2eg

## 问题

1. `zynqmp-a53-dfzu2eg` 平台驱动只有一个 uart 可以采用编译使用内建设备树添加的方式强制支持，需要额外支持 clk，pm-domain  驱动。
2.  `cvitek` 和 `phytium` 存在大量驱动，难以在短时间内快速支持 DM。
3. 由于目前 RISCV 架构完全不支持 DM，会导致 `qemu-virt64-riscv64` 不能使用后续即将提交的 virtio dm 重构版本，间接导致 `qemu-virt64-aarch64` 彻底支持 DM 可能耽误。

## 讨论

因此需要与社区成员协商目前方案（可能需要 **2～3 个月**完成）是否可以这样进行，如果可以，`cvitek` 和 `phytium` 平台需要如何快速支持，virtio 驱动问题是否暂时不允许  `qemu-virt64-riscv64` 使用。

> 在 RISCV 架构具体支持不属于当前 ISSUE 讨论内容

BernardXiong, 2025-01-09:
> 后续计划将在 aarch64 平台强制使用 DM

这些是否会和各个bsp相关，从而导致这部分强制支持到会比较犯难？

GuEe-GUI, 2025-01-09:
只会影响 aarch64 相关的 bsp，但是如果涉及到 bsp 存在多架构确实会造成一定的困难，因为其他架构还不支持 DM

GuEe-GUI, 2025-02-25:
zynqmp-a53-dfzu2eg 在 qemu（xlnx-zcu102）下测试功能正常：

[image]

GuEe-GUI, 2025-03-18:
cvitek，phytium 无法强制使用DM，故采用保留一份旧版本 libcpu 方式进行推进。
RISC-V 相关将退出一份 DM 版本 libcpu，并强制 qemu-virt64-riscv64 使用 DM 模式。

closed 2025-03-18 by GuEe-GUI