RT-Thread/rt-thread #9893: [bug] kill命令不能退出后台程序

yixinghua121, 2025-01-09:
### RT-Thread Version

c1166e0bf16aeda85e5facd01506973afcf6e8c6

### Hardware Type/Architectures

k230

### Develop Toolchain

GCC

### Describe the bug

1.后台执行一个程序：/bin/killme.elf &
2.查看进程的ID:list_process，找出刚后台执行应用的pid
3.kill掉后台进程：kill $pid -s 2
4.list_process查看现有的进程

heyuanjie87, 2025-01-21:
我这用kill 1是可以结束进程的，也可以用killall FILENAME https://github.com/RT-Thread/rt-thread/pull/9935

yixinghua121, 2025-03-05:
后台运行以下程序，发现后台程序没有运行起来

#include <stdio.h>
#include <rtthread.h>

int main()
{
    printf("Please kill me.\n");

    while (1)
    {
        rt_thread_mdelay(10000);
        printf("Please kill me.\n");
    }

    return 0;
}