RT-Thread/rt-thread #9896: [Bug] k230 utest PSE51部分测试失败

yixinghua121, 2025-01-09:
### RT-Thread Version

c1166e0bf16aeda85e5facd01506973afcf6e8c6

### Hardware Type/Architectures

k230

### Develop Toolchain

GCC

### Describe the bug

1.按照“步骤2”链接步骤，批量运行pse51测试例，部分失败；
2.步骤链接：https://support.rt-thread.com/canaan-creative/maix3/-/blob/master/software/userapps/testcases/pse51_test/README.md
3.代码链接：https://support.rt-thread.com/canaan-creative/maix3/-/tree/master/software/userapps/testcases/pse51_test
4.失败case：
other_tc/fork/3-1...FAILED
other_tc/fork/2-1...FAILED

pthread_tc/pthread_setschedparam/1-1...FAILED
pthread_tc/pthread_cancel/2-3...FAILED
pthread_tc/pthread_cancel/3-1...FAILED
pthread_tc/pthread_cancel/1-1...FAILED
pthread_tc/pthread_cancel/2-1...FAILED
pthread_tc/pthread_cancel/2-2...FAILED
pthread_tc/pthread_getschedparam/1-2...FAILED
pthread_tc/pthread_cond_signal/4-1...FAILED

sched_tc/sched_get_priority_max/2-1...FAILED
sched_tc/sched_get_priority_min/2-1...FAILED

signal_tc/sigqueue/7-1...FAILED
signal_tc/sigqueue/2-1...FAILED
signal_tc/killpg/6-1...FAILED
signal_tc/killpg/8-1...FAILED
signal_tc/killpg/5-1...FAILED

std_tc/unistd_h/pipe_tc...FAILED
std_tc/stdio_h/setbuf...FAILED

time_tc/clock_settime/6-1...FAILED
time_tc/clock_gettime/7-1...FAILED

heyuanjie87, 2025-01-22:
https://github.com/RT-Thread/rt-thread/pull/9943

heyuanjie87, 2025-01-22:
pipe_tc需要打开 components/libc下的#define RT_USING_POSIX_PIPE

msh />pipe_tc
msh />hello, world #1
hello, world #2
hello, world #3
{Test PASSED}

heyuanjie87, 2025-01-23:
pthread_tc/pthread_getschedparam/1-2失败原因为smart还不支持调整policy

heyuanjie87, 2025-01-23:
sched_tc/sched_get_priority_max/2-1:   https://github.com/bminor/musl/blob/master/src/sched/sched_get_priority_max.c
这个函数在musl中没设置errno

heyuanjie87, 2025-01-23:
pthread_tc/pthread_cancel/2-3：需要同步musl中pthread_cancel.c到罪行版
增加了如下代码
```
static void cancel_handler(int sig, siginfo_t *si, void *ctx)
{
	if (self->cancelasync) {
		pthread_sigmask(SIG_SETMASK, &uc->uc_sigmask, 0);
		__cancel();
	}
}
```
更改后qemu中测试成功
```
msh />cd pthread_cancel/
msh /pthread_cancel>2-3
msh /pthread_cancel>{Test PASSED}

msh /pthread_cancel>
```

heyuanjie87, 2025-01-24:
signal_tc/sigqueue:sigqueue接口在内核未实现

BernardXiong, 2025-01-25:
> pipe_tc需要打开 components/libc下的#define RT_USING_POSIX_PIPE
> 
> msh />pipe_tc msh />hello, world [#1](https://github.com/RT-Thread/rt-thread/pull/1) hello, world [#2](https://github.com/RT-Thread/rt-thread/pull/2) hello, world [#3](https://github.com/RT-Thread/rt-thread/pull/3) {Test PASSED}

这类应该是rt-smart必备的特性，可以在Kconfig中建立依赖关系

heyuanjie87, 2025-02-08:
std_tc/stdio_h/setbuf...FAILED: 我这暂未复现问题
```
msh />setbuf
msh />hello rt-smart

hello rt-smart

{Test PASSED}
```

heyuanjie87, 2025-02-08:
pthread_tc/pthread_cancel/3-1...FAILED：我这运行成功(已应用musl补丁，但补丁是修复异步取消，未跟踪是否有关联)
···
msh /pthread_cancel>3-1
msh /pthread_cancel>Thread cancelled after 1 ms.
{Test PASSED}
···

heyuanjie87, 2025-02-08:
pthread_tc/pthread_cancel/1-1...FAILED：测试成功
···
msh /pthread_cancel>1-1
msh /pthread_cancel>{Test PASSED}
···

heyuanjie87, 2025-02-08:
pthread_tc/pthread_cancel/2-1...FAILED：测试成功
```
msh /pthread_cancel>2-1
msh /pthread_cancel>{Test PASSED}
```

heyuanjie87, 2025-02-08:
pthread_tc/pthread_cancel/2-1...FAILED：测试成功
···
msh /pthread_cancel>2-2
msh /pthread_cancel>{Test PASSED}

···

heyuanjie87, 2025-02-08:
sched_tc/sched_get_priority_min/2-1...FAILED：此示例测试错误参数的返回值和errno，但musl未对此接口设置errno

heyuanjie87, 2025-02-08:
signal_tc/killpg/6-1...FAILED：https://github.com/RT-Thread/rt-thread/pull/9943

msh /killpg>5-1
msh /killpg>{Test PASSED}

heyuanjie87, 2025-02-08:
signal_tc/killpg/8-1...FAILED：https://github.com/RT-Thread/rt-thread/pull/9943

msh /killpg>8-1
msh /killpg>{Test PASSED}

heyuanjie87, 2025-02-08:
time_tc/clock_settime/6-1...FAILED：CLOCK_REALTIME需要RTC，系统未添加RTC
···
clock_gettime>7-1
msh /clock_gettime>CLOCK_MONOTONIC ret_tpset.tv_sec:33
CLOCK_MONOTONIC ret_tpset.tv_sec:34
[W/time] Cannot find a RTC device!
CLOCK_REALTIME ret_tpset.tv_sec:-274874464384
[W/time] Cannot find a RTC device!
[W/time] Cannot find a RTC device!
CLOCK_REALTIME ret_tpset.tv_sec:-274874464288
[W/time] Cannot find a RTC device!
[W/time] Cannot find a RTC device!
CLOCK_REALTIME ret_tpset.tv_sec:-274874464304
after sleep time not change or not right
···

heyuanjie87, 2025-02-08:
time_tc/clock_gettime/7-1...FAILED：CLOCK_MONOTONIC符合预期，CLOCK_REALTIME缺失RTC

```
clock_gettime>7-1
msh /clock_gettime>CLOCK_MONOTONIC ret_tpset.tv_sec:462
CLOCK_MONOTONIC ret_tpset.tv_sec:463
[W/time] Cannot find a RTC device!
CLOCK_REALTIME ret_tpset.tv_sec:-274874463920
[W/time] Cannot find a RTC device!
[W/time] Cannot find a RTC device!
CLOCK_REALTIME ret_tpset.tv_sec:3904453134506551659
[W/time] Cannot find a RTC device!
[W/time] Cannot find a RTC device!
CLOCK_REALTIME ret_tpset.tv_sec:-274874463856
after sleep time not change or not right
```

heyuanjie87, 2025-02-10:
other_tc/fork/2-1...FAILED：
此示例失败原因是获取到的sd_flag与设置的不同。
但是示例所用signo为SIGUSR1而下文提到SA_NOCLDSTOP只对SIGCHLD有效，所以rtt未保存此signo的flag。但在linux中是保存了的。
这不好直接判定为bug所以我暂时放弃修复此示例

https://www.man7.org/linux/man-pages/man2/sigaction.2.html
```
SA_NOCLDSTOP
              If signum is SIGCHLD, do not receive notification when
              child processes stop (i.e., when they receive one of
              SIGSTOP, SIGTSTP, SIGTTIN, or SIGTTOU) or resume (i.e.,
              they receive SIGCONT) (see [wait(2](https://www.man7.org/linux/man-pages/man2/wait.2.html)
```

heyuanjie87, 2025-02-10:
signal_tc/killpg/6-1...FAILED：https://github.com/RT-Thread/rt-thread/pull/9981

```
msh /killpg>6-1
msh /killpg>{Test PASSED}

msh /killpg>6-1
msh /killpg>{Test PASSED}

msh /killpg>6-1
msh /killpg>{Test PASSED}

msh /killpg>6-1
msh /killpg>{Test PASSED}
```

closed 2025-02-18 by Rbb666