RT-Thread/rt-thread #9897: [Bug] k230 utest libc测试部分测试例失败

yixinghua121, 2025-01-09:
### RT-Thread Version

c1166e0bf16aeda85e5facd01506973afcf6e8c6

### Hardware Type/Architectures

k230

### Develop Toolchain

GCC

### Describe the bug

1.测试libc，部分case失败；
2.代码链接：https://support.rt-thread.com/canaan-creative/maix3/-/tree/master/software/userapps/testcases/test
3.失败case：
utest.pingpong_test.pingpong_timeout_tc
utest.libc_test.file.fcntl
utest.libc_test.file.fscanf
utest.libc_test.math.powf

heyuanjie87, 2025-01-22:
pipe_tc需要打开 components/libc下的#define RT_USING_POSIX_PIPE

```
msh />pipe_tc
msh />hello, world #1
hello, world #2
hello, world #3
{Test PASSED}
```

heyuanjie87, 2025-03-21:
fscanf的问题是报的这个错吗,需要创建一个可写的sdcard目录。测试代码创建了文件："char *temp_file = "/sdcard/temp_file_fscanf";"
```
msh />fscanf.elf
msh />fopen tmpfile failed
fscanf.c:65: !!(f=writetemp("      42")) failed (failed to make temp file)
fopen tmpfile failed
fscanf.c:76: !!(f=writetemp("[abc123]....x")) failed (failed to make temp file)
fopen tmpfile failed
fscanf.c:90: !!(f=writetemp("0x1p 12")) failed (failed to make temp file)
fopen tmpfile failed
fscanf.c:114: !!(f=writetemp("0x.1p4    012")) failed (failed to make temp file)
fopen tmpfile failed
fscanf.c:127: !!(f=writetemp("0xx")) failed (failed to make temp file)
```

yixinghua121, 2025-04-01:
utest.libc_test.file.fscanf, 问题已经解决；