RT-Thread/rt-thread #9900: [Bug] dfs_v2打开pagecache，运行elf后会存在memleak

wycwyhwyq, 2025-01-09:
### RT-Thread Version

53bd56ccf9cb77dd1cec3c3c9e1fde24b0a95267

### Hardware Type/Architectures

risc-v/K230

### Develop Toolchain

GCC

### Describe the bug

1. dfs_v2打开pagecache，运行任意elf后，通过free对比运行前后发现存在memleak

### Other additional context

定位到具体位置有两处：
1. 加载elf文件时对应的`struct dfs_file *file`结构未释放，原因为elf map时会增加`file->ref_count`，导致执行`close->fd_release->fdt_fd_release`时`file->ref_count`不等于1，未能正常释放
2. 用户程序退出时会执行`lwp_exit->_clear_child_tid->sys_futex`，此处是个share futex，不会被删除

对于share futex，还有一个疑问，对于没有进程使用的futex似乎没有删除操作？

polarvid, 2025-01-10:
share futex 设计时有 lazy 任务定期清理。不过由于 libc 的设计算法，整个系统生命周期中它的 footpirnt 基本上和 tick 形成  O(1) 增长关系。所以这个研发任务定为”必要不紧急“。因为重要任务很多，所以搁置了。如果题主有兴趣可以 PR 补充。

BernardXiong, 2025-01-12:
> share futex 设计时有 lazy 任务定期清理。不过由于 libc 的设计算法，整个系统生命周期中它的 footpirnt 基本上和 tick 形成 O(1) 增长关系。所以这个研发任务定为”必要不紧急“。因为重要任务很多，所以搁置了。如果题主有兴趣可以 PR 补充。

是否可以简单展开说明下？

heyuanjie87, 2025-01-14:
确实存在内存泄漏

1. 打印出mmap可执行文件的file指针
[image]

[image]

2.  打印rt_free的所有指针
[image]

3. 把程序退出时的日志保存下来搜索0xffffffc0003a19f8，只出现了一次
[image]

heyuanjie87, 2025-01-14:
确实存在内存泄漏

1. 打印执行文件时创建的file指针
[image]

2. 打印rt_free的指针
[image]

3. 在程序退出时输出的日志中查找file指针，只有一条
[image]

Rbb666, 2025-01-17:
关联：https://github.com/RT-Thread/rt-thread/pull/9917

Rbb666, 2025-02-09:
@wycwyhwyq https://github.com/RT-Thread/rt-thread/pull/9917 已合并，可以再测试下