RT-Thread/rt-thread #9905: [Feature] scons支持更多命令以提供bsp/工程更多背景信息

BernardXiong, 2025-01-12:
### Describe problem solved by the proposed feature

目前`scons --target=vsc`方式可以支持提供部分bsp/工程信息，例如RTT_ROOT，Groups等，但从工程角度出发，还是缺少部分，希望可以有更多的信息以json文本方式提供出来。也许这些信息是由bsp维护人员提供。

### Describe your preferred solution

由bsp维护人来提供bsp/工程信息，以json文本方式。需要包括：

- 可以支持哪些工具链，例如gcc，llvm，iar或其他
- 对应工具链的链接脚本文件是哪个；

也可以对`rtconfig.py`文件做重构，一些信息可以从这个json文本中直接获得，而不是再通过`rtconfig.py`文件指定。

或者未来可以让`rtconfig.py`文件更json文本化。

### Describe possible alternatives

在json文件的基础上，未来可以通过带UI <字符界面UI，或图形化UI>的配置工具。

supperthomas, 2025-01-13:
非常赞同该提案。形式的话，我建议yaml文件，看上去比json文件更直观一些，也方便修改。图形化UI也方便
可以参考studio的yaml文件。
https://github.com/RT-Thread-Studio/sdk-bsp-stm32f407-spark/blob/main/sdk-bsp-rt-spark.yaml

BernardXiong, 2025-01-14:
> 非常赞同该提案。形式的话，我建议yaml文件，看上去比json文件更直观一些，也方便修改。图形化UI也方便 可以参考studio的yaml文件。 https://github.com/RT-Thread-Studio/sdk-bsp-stm32f407-spark/blob/main/sdk-bsp-rt-spark.yaml

图形化UI，是不是还是json更方便？我争取这周末能推新的vscode扩展版本