RT-Thread/rt-thread #9906: [Bug] Should fix the checking error in ci for rt-smart building.

BernardXiong, 2025-01-12:
### RT-Thread Version

c5a79de38e400f5449c3eabb03739e8d3007283a

### Hardware Type/Architectures

generic

### Develop Toolchain

GCC

### Describe the bug

当前在ci中使用cppcheck检查时，如开启了smart，可能会报未使能dfs的错误（因为rt-smart强制关联了dfs），需要确保在cppcheck状态下也能通过。

例如这里的错误：
https://github.com/RT-Thread/rt-thread/actions/runs/12673562082/job/35320219895?pr=9818

BernardXiong, 2025-01-12:
@zhuzhuzhus please fix this issue, thanks.

supperthomas, 2025-01-13:
这个error其实不需要修改，本质上是cppcheck没有检测到该宏导致的，
如果要强行给cppcheck加该宏的话，可以在下面的文件中添加
https://github.com/RT-Thread/rt-thread/blob/c5a79de38e400f5449c3eabb03739e8d3007283a/tools/ci/cpp_check.py#L53-L72

zhuzhuzhus, 2025-01-14:
****> 这个error其实不需要修改，本质上是cppcheck没有检测到该宏导致的， 如果要强行给cppcheck加该宏的话，可以在下面的文件中添加
> 
> https://github.com/RT-Thread/rt-thread/blob/c5a79de38e400f5449c3eabb03739e8d3007283a/tools/ci/cpp_check.py#L53-L72

加个对lwp.c 的判断应该就行了吧：
```python
        macros = []
        if os.path.basename(file) == 'lwp.c':
            macros.append('-DRT_USING_DFS')  

        result = subprocess.run(
            [
                'cppcheck',
                '-DRT_ASSERT(x)=',
                '-DRTM_EXPORT(x)=',
                '-Drt_list_for_each_entry(a,b,c)=a=(void*)b;',
                '-I include',
                '-I thread/components/finsh',
                '--suppress=syntaxError',
                '--check-level=exhaustive',
                '--enable=warning',
                'performance',
                'portability',
                '--inline-suppr',
                '--error-exitcode=1',
                '--force',
                file
            ] + macros, 
```

zhuzhuzhus, 2025-01-14:
> 这个error其实不需要修改，本质上是cppcheck没有检测到该宏导致的， 如果要强行给cppcheck加该宏的话，可以在下面的文件中添加
> 
> https://github.com/RT-Thread/rt-thread/blob/c5a79de38e400f5449c3eabb03739e8d3007283a/tools/ci/cpp_check.py#L53-L72

加个对lwp.c 的判断应该就行了吧：

        macros = []
        if os.path.basename(file) == 'lwp.c':
            macros.append('-DRT_USING_DFS')  

        result = subprocess.run(
            [
                'cppcheck',
                '-DRT_ASSERT(x)=',
                '-DRTM_EXPORT(x)=',
                '-Drt_list_for_each_entry(a,b,c)=a=(void*)b;',
                '-I include',
                '-I thread/components/finsh',
                '--suppress=syntaxError',
                '--check-level=exhaustive',
                '--enable=warning',
                'performance',
                'portability',
                '--inline-suppr',
                '--error-exitcode=1',
                '--force',
                file
            ] + macros,

BernardXiong, 2025-01-14:
fixed with #9912

closed 2025-01-14 by BernardXiong