RT-Thread/rt-thread #9909: [Feature] bsp/cvitek 控制台启动打印区分大小核以及 RISC-V 和 AARCH

unicornx, 2025-01-13:
### Describe problem solved by the proposed feature

duo 的 cou 组合比较复杂：

| BSP 名称      | 大小核  | 芯片架构        | 默认串口控制台 | 备注     |
| ------------- | ------- |---------------- | -------------- | -------- |
| cv18xx_risc-v | 大核    | RISC-V C906     | uart0          | 支持 MMU，支持 RT-Thread 标准版 和 RT-SMART 模式，默认运行 RT-SMART 版本 |
| c906-little   | 小核    | RISC-V C906     | uart1          | 无 MMU，运行 RT-Thread 标准版 |
| cv18xx_aarch64| 大核    | ARM Cortex A53  | uart0          | 支持 MMU， 支持 RT-Thread 标准版 和 RT-SMART 版，默认运行 RT-Thread 标准版本 |

启动打印时如果能区分一下，对于测试会比较好，可以明确知道对应的 cpu 是不是运行正常。
目前无区分，只能知道时 RT-smart 还是标准版

BernardXiong, 2025-01-14:
smart版本把procfs加上了，后面可以直接 `cat /proc/cpu`

unicornx, 2025-01-16:
> smart版本把procfs加上了，后面可以直接 `cat /proc/cpu`

这个 issue 的想法是直接在启动打印中区分，方便测试和开发过程中明确串口连接正常，以及是否运行的是大核还是小核的版本。

PR #9919 (2025-01-16): bsp/cvitek: print arch info. during boot-up
Fixed #9909 

Duo's CPU combination is more complicated:

| BSP           | B/L core| ISA             | UART  |
| ------------- | ------- |---------------- |-------|
| cv18xx_risc-v | Big     | RISC-V C906     | UART0 |
| c906-little   | Littel  | RISC-V C906     | UART1 |
| cv18xx_aarch64| Big     | ARM Cortex A53  | UART0 |

Printing ISA and big and small core information
during the boot process helps developers/testers
determine the CPU and serial port corresponding to the current console.

In addition, the RTT logo printing has already
distinguis [... cut]
closed 2025-01-16 by supperthomas