RT-Thread/rt-thread #9931: [Bug] LVGL package c99 in keil

supperthomas, 2025-01-18:
### RT-Thread Version

master

### Hardware Type/Architectures

rt-spark

### Develop Toolchain

MDK Keil

### Describe the bug

ArmClang: error: unsupported option '--c99'
ArmClang: error: unsupported option '--gnu'; did you mean '-fgnu'?

### Other additional context

packages\LVGL-v8.3.11\env_support\rt-thread\SConscript

if rtconfig.PLATFORM == 'gcc' or rtconfig.PLATFORM == 'armclang': # GCC or Keil AC6
    LOCAL_CFLAGS += ' -std=c99'
elif rtconfig.PLATFORM == 'armcc': # Keil AC5
    LOCAL_CFLAGS += ' --c99 --gnu'

Keil AC6，这里没有能识别到。

supperthomas, 2025-01-18:
感觉要加个命令， `scons --target=mdk5-AC6`

mysterywolf, 2025-01-18:
若使用AC6 我记得rtconfig.py需要调整，将platform改一下 你得问问郭老师

supperthomas, 2025-01-19:
> 若使用AC6 我记得rtconfig.py需要调整，将platform改一下 你得问问郭老师

没看到文档

supperthomas, 2025-01-19:
@mysterywolf rtduino里面也有，你在KEIL AC6下面试过吗？能跑吗？

supperthomas, 2025-01-19:
ArmClang: error: unsupported option '--gnu'; did you mean '-fgnu'?

supperthomas, 2025-01-19:
[image]

supperthomas, 2025-01-19:
[image]

supperthomas, 2025-01-19:
rtconfig.py 如下修改即可
```
elif CROSS_TOOL == 'keil':
    PLATFORM    = 'armclang'
    EXEC_PATH   = r'C:/Keil_v5'
```

closed 2025-01-19 by supperthomas