RT-Thread/rt-thread #9937: [Feature] ci/cd 在qemu上实现方案

supperthomas, 2025-01-20:
### Describe problem solved by the proposed feature

可以在原有的ci.attachconfig.yml中实现如下规则：

```yml
bsp_board_info:
    arch: arm
    toolchain: arm-none-eabi-gcc
    pre_build:
       ./tools/ci/toolchain.sh arm-none-eabi-gcc
      apt-get install qemu-system
    post_build: objcopy rtthread.elf rtthread.hex
    run_cmd: ./qemu-nographic.sh
    qemu_flag: true
pkg.tools.coremark:
  kconfig: 
    - CONFIG_PKG_USING_COREMARK=y
  ci_build_run_flag : true
  checkresult: '[==========] [ utest ] end'
```

### Describe your preferred solution

```yml
bsp_board_info:
    arch: arm
    toolchain: arm-none-eabi-gcc
    pre_build:
       ./tools/ci/toolchain.sh arm-none-eabi-gcc
      apt-get install qemu-system
    post_build: objcopy rtthread.elf rtthread.hex
    run_cmd: ./qemu-nographic.sh
    qemu_flag: true
pkg.tools.coremark:
  kconfig: 
    - CONFIG_PKG_USING_COREMARK=y
  ci_build_run_flag : true
  checkresult: '[==========] [ utest ] end'
```

supperthomas, 2025-01-20:
@kurisaW

kurisaW, 2025-01-26:
@supperthomas 这个之前和满老师讨论过，您的attachconfig是面向整个bsp编译的，然后qemu的auto utest 脚本实际只需要针对那三个qemu平台进行配置，分开更方便维护一些，然后配置信息我们放在.github/utest下

supperthomas, 2025-01-26:
> [@supperthomas](https://github.com/supperthomas) 这个之前和满老师讨论过，您的attachconfig是面向整个bsp编译的，然后qemu的auto utest 脚本实际只需要针对那三个qemu平台进行配置，分开更方便维护一些，然后配置信息我们放在.github/utest下

可以呀，这个后面当拓展。

supperthomas, 2025-01-26:
@Li-Dongze