RT-Thread/rt-thread #9941: [Feature] ci.attachconfig.yml 和scons结合使用

supperthomas, 2025-01-21:
### Describe problem solved by the proposed feature

为了方便用户方便使用yml中的配置，添加如下命令
`scons --attach=? ` 打印出yml中的所有配置
参考如下：
[image]

`scons --attach=component.cherryusb_cdc` 该命令把yml中的component.cherryusb_cdc配置放到.config的最后中，并且执行一下menuconfig

`scons --attach=default`  执行`git reset` 命令恢复.config为原样。

### Describe your preferred solution

如有其他建议可以comment

hydevcode, 2025-01-21:
好神奇，pr中提到这个issues，就会自动在这边显示

hydevcode, 2025-01-26:
[RT-Thread-【RT-Thread】【ci】【scons】将ci.attachconfig.yml和scons结合使用 - RT-Thread](https://club.rt-thread.org/ask/article/d273bbcd1f8779bc.html)
写了一篇关于这个功能的帖子，感兴趣的可以看看

supperthomas, 2025-02-21:
https://mp.weixin.qq.com/s/CAu4Nyiq4-Of78il-yaxwg

Yunlk, 2026-10-20:
## 核实结论：`scons --attach` 已实现且实测可用，建议关闭本 issue

master `1c801efa` 实测。因为只有 `bsp/renesas/*` 下带 `attachconfig.yml`，用的是 `bsp/renesas/ra6m3-hmi-board`。

issue 正文提的三条命令逐条验证：

**1) `scons --attach=?` 列出所有配置**

```
$ scons --attach=?
✅ AttachConfig has:
 ------system------
system.nano
 ------devices------
devices.gpio
devices.uart
devices.soft_i2c
```

**2) `scons --attach=<name>` 把配置写进 `.config` 并执行配置**

```
$ scons --attach=devices.uart
✅ AttachConfig add success!
→ .config 追加 CONFIG_BSP_USING_UART=y，同时生成 .config.origin / rtconfig.h.origin 备份
```

再执行 `--attach=devices.gpio` 可以叠加，符合预期。

**3) `scons --attach=default` 恢复原样**

```
$ scons --attach=default
✅ Default .config and rtconfig.h recovery success!
→ 备份被还原并删除，git status 显示 .config 无变更
```

实现位置对应关系：`tools/options.py` 注册 `--attach`，`tools/building.py:395` 调用 `GenAttachConfigProject()`，`tools/attachconfig.py` 实现，`ChangeLog.md` 有记录。

### 两点补充

1. `devices.strict` 不会出现在 `--attach=?` 的列表里。原因是 `GetAttachConfig()` 会跳过没有 `kconfig` 键的条目（`attachconfig.py:78-79`），而 `devices.strict` 只有 `depend_scons_arg`。当它被 `devices.gpio` 依赖时，用户在列表里看不到它，容易误以为不存在。不算 bug，但可以考虑在列表里标注「仅依赖/无 kconfig」。

2. 中文 Windows 默认 GBK 控制台下，`--attach=?` 会直接抛 `UnicodeEncodeError`（`attachconfig.py:18` 的 ✅）。这是独立问题，我另开了一个 issue。

如果没有其他待办，建议关闭本 issue。