RT-Thread/rt-thread #9964: [Bug][serial_v1] 主频较高时可能会导致发送不完整

kaidegit, 2025-02-03:
### RT-Thread Version

master

### Hardware Type/Architectures

H730 with internal flash

### Develop Toolchain

MDK Keil

### Describe the bug

在主频550M，打开优化时，使用rt_kprintf只能打出首字符。

调查发现在stm32的`serial_v1`驱动中，会使用一个较为硬编码（由库中define）的值检查是否发送超时。在主频较高且波特率较低时可能会触发。
https://github.com/RT-Thread/rt-thread/blob/0cc4dc4a3dd5bbb3b95ffdc7cd7f4e740a9c6b23/bsp/stm32/libraries/HAL_Drivers/drivers/drv_usart.c#L344-L361
而`rt_kpinrtf`调用`_kputs`发送时未处理`rt_device_write`的返回值，没有重发。
https://github.com/RT-Thread/rt-thread/blob/0cc4dc4a3dd5bbb3b95ffdc7cd7f4e740a9c6b23/src/kservice.c#L312-L335

相关PR：https://github.com/RT-Thread/rt-thread/pull/9543

这个PR把代码改正确了，但是导致了这个问题 。🤦‍♂️

supperthomas, 2025-02-05:
这个是不是要考虑把TX_BLOCK_TIMEOUT 这个放到Kconfig中，给不同平台适配？

kaidegit, 2025-02-05:
我觉得既然是stm32 driver里的适配问题，拿hal的api获取一下主频算一下就好了

supperthomas, 2025-02-05:
> 我觉得既然是stm32 driver里的适配问题，拿hal的api获取一下主频算一下就好了

欢迎PR修复一下。

kaidegit, 2025-02-05:
主要是有cache、flash等待、art加速器这些因素在我怕算不准😂

supperthomas, 2025-02-05:
> 主要是有cache、flash等待、art加速器这些因素在我怕算不准😂

那就加个Kconfig，每个stm32 自己配一下，默认一个最大值。

supperthomas, 2025-02-05:
不过v1是不是用的不多了？都要转V2了？可以试试v2

kaidegit, 2025-02-05:
主要看默认config都是v1

CYFS3, 2025-02-05:
> > 主要是有cache、flash等待、art加速器这些因素在我怕算不准😂
> 
> 那就加个Kconfig，每个stm32 自己配一下，默认一个最大值。

有下面几个方案，看一下哪个合适
* 可以直接像其他mcu适配方案一样直接，死等发送标志位完成，我看stm32 v2也是这样做的
* 每款stm32加个kconfig，调整合适的值
* 根据mcu的频率和波特率计算适合的值，因为我之前在f1上面遇到的情况是9600会超时而已，其他的波特率能够正常发送

wdfk-prog, 2025-02-17:
> > > 主要是有cache、flash等待、art加速器这些因素在我怕算不准😂
> > 
> > 
> > 那就加个Kconfig，每个stm32 自己配一下，默认一个最大值。
> 
> 有下面几个方案，看一下哪个合适
> 
> * 可以直接像其他mcu适配方案一样直接，死等发送标志位完成，我看stm32 v2也是这样做的
> * 每款stm32加个kconfig，调整合适的值
> * 根据mcu的频率和波特率计算适合的值，因为我之前在f1上面遇到的情况是9600会超时而已，其他的波特率能够正常发送

死等可能导致一直阻塞,这样的代码并不合理.
通过计算并添加余量吧.可以提供API由外边修改timeout

kaidegit, 2025-02-23:
附上v2的驱动，没关注超时，emmm
https://github.com/RT-Thread/rt-thread/blob/0e9b7d9553447272da214d3284bfed906b2f90f5/bsp/stm32/libraries/HAL_Drivers/drivers/drv_usart_v2.c#L307-L318

CYFS3, 2025-02-23:
> 附上v2的驱动，没关注超时，emmm
> 
> [rt-thread/bsp/stm32/libraries/HAL_Drivers/drivers/drv_usart_v2.c](https://github.com/RT-Thread/rt-thread/blob/0e9b7d9553447272da214d3284bfed906b2f90f5/bsp/stm32/libraries/HAL_Drivers/drivers/drv_usart_v2.c#L307-L318)
> 
> Lines 307 to 318 in [0e9b7d9](/RT-Thread/rt-thread/commit/0e9b7d9553447272da214d3284bfed906b2f90f5)
> 
>  static int stm32_putc(struct rt_serial_device *serial, char c) 
>  { 
>      struct stm32_uart *uart; 
>      RT_ASSERT(serial != RT_NULL); 
>   
>      uart = rt_container_of(serial, struct stm32_uart, serial); 
>      UART_INSTANCE_CLEAR_FUNCTION(&(uart->handle), UART_FLAG_TC); 
>      UART_SET_TDR(&uart->handle, c); 
>      while (__HAL_UART_GET_FLAG(&(uart->handle), UART_FLAG_TC) == RESET); 
>   
>      return 1; 
>  }

确实很多实现都是死等的

kaidegit, 2025-02-23:
个人觉得，想简单些，等1个tick得了，就算开到20k的tick，50ms也足够2400波特率发出十个了。或者直接学v2删删掉。不是没开外设时钟一般不会卡这的，能卡在那反而方便排查问题。

wdfk-prog, 2025-02-24:
> 个人觉得，想简单些，等1个tick得了，就算开到20k的tick，50ms也足够2400波特率发出十个了。或者直接学v2删删掉。不是没开外设时钟一般不会卡这的，能卡在那反而方便排查问题。

但是如果是关注整体系统的话,任何的死循环都是不能接受的

closed 2025-03-03 by supperthomas

CYFS3, 2025-03-03:
#10022 可以通过menuconfig设置超时的值

PR #10022 (2025-03-03): [bsp][stm32][uart]:fix stm32 uart timout
## 拉取/合并请求描述：(PR description)

[

#### 为什么提交这份PR (why to submit this PR)
fixed: #9964 

#### 你的解决方案是什么 (what is your solution)

#### 请提供验证的bsp和config (provide the config and bsp) 

- BSP:

- .config:

- action:

]