RT-Thread/rt-thread #9968: [Feature] bsp/cvitek/cv18xx_aarch64 move prebuilds to rttpkgtool

unicornx, 2025-02-04:
### Describe problem solved by the proposed feature

目前 rttpkgtool 替换了原来的 cvitek_bootloader, 但只覆盖了 riscv 部分（大核和小核），但是不包括对 aarch64 的大核部分。

考虑在 rttpkgtool 中也支持 aarch64 大核，这样所有的prebuild 部分都可以从 RTT 主仓移除。

jojoandgyc, 2025-02-05:
ccept

jojoandgyc, 2025-02-18:
汪老师，您看这样如何

[image]

，链接地址 https://github.com/plctlab/rttpkgtool/commit/c10cae99ecd7e9766f9e16e41305ab17327dfc3a

unicornx, 2025-02-19:
> 地址 [plctlab/rttpkgtool@c10cae9](https://github.com/plctlab/rttpkgtool/commit/c10cae99ecd7e9766f9e16e41305ab17327dfc3a)

PR #10151 (2025-03-29): bsp: cvitek: use rttpkgtool for aarch64
Fixed  #9968
link #10129 #10105 

1.Using scripts in rttpkgtool to make firmware.
2.Deleted duplicate prebuild files.
3.Modify README.md

## 拉取/合并请求描述：(PR description)

[

#### 为什么提交这份PR (why to submit this PR)

#### 你的解决方案是什么 (what is your solution)

#### 请提供验证的bsp和config (provide the config and bsp) 

- BSP:

- .config:

- action:

]

<
commit 15ef3674 (2025-03-29): bsp: cvitek: use rttpkgtool for aarch64

Fixed  #9968

1.Using scripts in rttpkgtool to make firmware.
2.Deleted duplicate prebuild files.
3.Modify README.md
closed 2025-03-29 by Rbb666