RT-Thread/rt-thread #9992: [Bug] qemu-risc64 运行信号demo会报错。

jiuyueshenhua, 2025-02-14:
### RT-Thread Version

9afe6a51826fb84de229de220dab2077aa5734b2

### Hardware Type/Architectures

qemu-virt64-riscv

### Develop Toolchain

GCC

### Describe the bug

https://www.rt-thread.org/document/site/#/rt-thread-version/rt-thread-standard/programming-manual/ipc2/ipc2?id=%E4%BF%A1%E5%8F%B7
在qemu-vir64-riscv的环境下执行信号使用例程，会出现以下报错：
看样子似乎是libcpu的支持有问题？
catcatbing@catcatbing-thinkpad-l15-gen-4:~/dev/project/join_rtt/rt-thread/bsp/qemu-virt64-riscv$ ./qemu-nographic.sh 

OpenSBI v1.3
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|___/_____|
        | |
        |_|

Platform Name             : riscv-virtio,qemu
Platform Features         : medeleg
Platform HART Count       : 1
Platform IPI Device       : aclint-mswi
Platform Timer Device     : aclint-mtimer @ 10000000Hz
Platform Console Device   : uart8250
Platform HSM Device       : ---
Platform PMU Device       : ---
Platform Reboot Device    : sifive_test
Platform Shutdown Device  : sifive_test
Platform Suspend Device   : ---
Platform CPPC Device      : ---
Firmware Base             : 0x80000000
Firmware Size             : 322 KB
Firmware RW Offset        : 0x40000
Firmware RW Size          : 66 KB
Firmware Heap Offset      : 0x48000
Firmware Heap Size        : 34 KB (total), 2 KB (reserved), 9 KB (used), 22 KB (free)
Firmware Scratch Size     : 4096 B (total), 760 B (used), 3336 B (free)
Runtime SBI Version       : 1.0

Domain0 Name              : root
Domain0 Boot HART         : 0
Domain0 HARTs             : 0*
Domain0 Region00          : 0x0000000002000000-0x000000000200ffff M: (I,R,W) S/U: ()
Domain0 Region01          : 0x0000000080040000-0x000000008005ffff M: (R,W) S/U: ()
Domain0 Region02          : 0x0000000080000000-0x000000008003ffff M: (R,X) S/U: ()
Domain0 Region03          : 0x0000000000000000-0xffffffffffffffff M: (R,W,X) S/U: (R,W,X)
Domain0 Next Address      : 0x0000000080200000
Domain0 Next Arg1         : 0x000000008fe00000
Domain0 Next Mode         : S-mode
Domain0 SysReset          : yes
Domain0 SysSuspend        : yes

Boot HART ID              : 0
Boot HART Domain          : root
Boot HART Priv Version    : v1.12
Boot HART Base ISA        : rv64imafdch
Boot HART ISA Extensions  : time,sstc
Boot HART PMP Count       : 16
Boot HART PMP Granularity : 4
Boot HART PMP Address Bits: 54
Boot HART MHPM Count      : 16
Boot HART MIDELEG         : 0x0000000000001666
Boot HART MEDELEG         : 0x0000000000f0b509
heap: [0x80299070 - 0x84299070]

 \ | /
- RT -     Thread Operating System
 / | \     5.2.0 build Feb 14 2025 16:27:32
 2006 - 2024 Copyright by RT-Thread team
lwIP-2.0.3 initialized!
[I/sal.skt] Socket Abstraction Layer initialize success.
[I/utest] utest is initialize success.
[I/utest] total utest testcase num: (0)
file system initialization done!
Hello RISC-V
Hello from main file!
msh />signalt
thread1 count : 0
thread1 count : 1
thread1 count : 2
msh />thread1 received signal 30
thread1 count : 3
Function[_thread_sleep]: scheduler is not available
(0) assertion failed at function:_thread_sleep, line number:635 
Unhandled Exception 5:Load Access Fault
scause:0x0000000000000005,stval:0x0000000000000010,sepc:0x0000000080242c60
--------------Dump Registers-----------------
Function Registers:
        ra(x1) = 0x0000000080242d60     user_sp = 0x00000000803a4de0
        gp(x3) = 0x0000000080273128     tp(x4) = (nil)
Temporary Registers:
        t0(x5) = 0x000000008025dc8a     t1(x6) = 0x0000000000000003
        t2(x7) = 0x00000000802c27d8
        t3(x28) = 0x0000000000000007    t4(x29) = (nil)
        t5(x30) = (nil) t6(x31) = (nil)
Saved Registers:
        s0/fp(x8) = 0x00000000803a4e50  s1(x9) = 0x000000008025f6a2
        s2(x18) = (nil) s3(x19) = (nil)
        s4(x20) = (nil) s5(x21) = (nil)
        s6(x22) = (nil) s7(x23) = (nil)
        s8(x24) = (nil) s9(x25) = (nil)
        s10(x26) = (nil)        s11(x27) = (nil)
Function Arguments Registers:
        a0(x10) = 0x000000008027b2a8    a1(x11) = 0x00000000803a4f10
        a2(x12) = 0x00000000803a4e6c    a3(x13) = 0x0000000000000041
        a4(x14) = (nil) a5(x15) = 0x0000000000000010
        a6(x16) = (nil) a7(x17) = 0x0000000000004000
sstatus = 0x8000000200046120
        Supervisor Interrupt Disabled
        Last Time Supervisor Interrupt Enabled
        Last Privilege is Supervisor Mode
        Permit to Access User Page
        Not Permit to Read Executable-only Page
satp = (nil)
        Current Page Table(Physical) = (nil)
        Current ASID = (nil)
        Mode = No Address Translation/Protection Mode
-----------------Dump OK---------------------
--------------Thread list--------------
current thread: thread1
--------------Backtrace--------------
please use: addr2line -e rtthread.elf -a -f 0x80242c60Unhandled Exception 5:Load Access Fault
[E/libcpu.trap] 
-------- [SEVER ERROR] --------
[E/libcpu.trap] Nested trap detected
[E/libcpu.trap] scause:0x0000000000000005,stval:0x0000000000000010,sepc:0x0000000080242c60

--------------Dump Registers-----------------
Function Registers:
        ra(x1) = 0x0000000080242d60     user_sp = 0x00000000803a4a40
        gp(x3) = 0x0000000080273128     tp(x4) = (nil)
Temporary Registers:
        t0(x5) = 0x0000000000000005     t1(x6) = 0x0000000000000008
        t2(x7) = 0x00000000802c27d8
        t3(x28) = 0x0000000000000007    t4(x29) = (nil)
        t5(x30) = (nil) t6(x31) = (nil)
Saved Registers:
        s0/fp(x8) = 0x00000000803a4ab0  s1(x9) = 0x000000008025f6a2
        s2(x18) = (nil) s3(x19) = (nil)
        s4(x20) = (nil) s5(x21) = (nil)
        s6(x22) = (nil) s7(x23) = (nil)
        s8(x24) = (nil) s9(x25) = (nil)
        s10(x26) = (nil)        s11(x27) = (nil)
Function Arguments Registers:
        a0(x10) = 0x000000008027b2a8    a1(x11) = 0x00000000803a4e50
        a2(x12) = 0x00000000803a4acc    a3(x13) = 0x000000000000000b
        a4(x14) = (nil) a5(x15) = 0x0000000000000010
        a6(x16) = (nil) a7(x17) = 0x0000000000000200
sstatus = 0x0000000200044100
        Supervisor Interrupt Disabled
        Last Time Supervisor Interrupt Disabled
        Last Privilege is Supervisor Mode
        Permit to Access User Page
        Not Permit to Read Executable-only Page
satp = (nil)
        Current Page Table(Physical) = (nil)
        Current ASID = (nil)
        Mode = No Address Translation/Protection Mode
-----------------Dump OK---------------------
shutdown...
catcatbing@catcatbing-thinkpad-l15-gen-4:~/dev/project/join_rtt/rt-thread/bsp/qemu-virt64-riscv$ addr2line -e rtthread.elf -a -f 0x80242c60Unhandled
0x0000000080242c60
_query
/home/catcatbing/dev/project/join_rtt/rt-thread/libcpu/risc-v/common64/mmu.c:366

unicornx, 2025-02-19:
> 在qemu-vir64-riscv的环境下执行信号使用例程

具体例程是什么？给出的链接中没有看到。

jiuyueshenhua, 2025-02-23:
文档中心的线程间通信章节的信号例程：

```C++
#include <rtthread.h>

#define THREAD_PRIORITY         25
#define THREAD_STACK_SIZE       512
#define THREAD_TIMESLICE        5

static rt_thread_t tid1 = RT_NULL;

/* 线程 1 的信号处理函数 */
void thread1_signal_handler(int sig)
{
    rt_kprintf("thread1 received signal %d\n", sig);
}

/* 线程 1 的入口函数 */
static void thread1_entry(void *parameter)
{
    int cnt = 0;

    /* 安装信号 */
    rt_signal_install(SIGUSR1, thread1_signal_handler);
    rt_signal_unmask(SIGUSR1);

    /* 运行 10 次 */
    while (cnt < 10)
    {
        /* 线程 1 采用低优先级运行，一直打印计数值 */
        rt_kprintf("thread1 count : %d\n", cnt);

        cnt++;
        rt_thread_mdelay(100);
    }
}

/* 信号示例的初始化 */
int signal_sample(void)
{
    /* 创建线程 1 */
    tid1 = rt_thread_create("thread1",
                            thread1_entry, RT_NULL,
                            THREAD_STACK_SIZE,
                            THREAD_PRIORITY, THREAD_TIMESLICE);

    if (tid1 != RT_NULL)
        rt_thread_startup(tid1);

    rt_thread_mdelay(300);

    /* 发送信号 SIGUSR1 给线程 1 */
    rt_thread_kill(tid1, SIGUSR1);

    return 0;
}

/* 导出到 msh 命令列表中 */
MSH_CMD_EXPORT(signal_sample, signal sample);

```

eatvector, 2025-07-24:
已解决，可关闭