RT-Thread/rt-thread #9994: [Feature] 建立一种机制，能否方便地获取两个特定版本之间内核 API 的 diff 列表

unicornx, 2025-02-16:
### Describe problem solved by the proposed feature

目前在发布版本时，无法方便获取两个版本之间 kernel API 的改变，如果有这个 change list，依赖的软件包也能知道如何修改。

目前基于 doxygen 在整理 kernal 的 API 文档信息，这个工作可以作为一个副产品做起来。

Rbb666, 2025-02-16:
目前的一些内核相关的changelog在文档中心有一份：https://www.rt-thread.org/document/site/#/rt-thread-version/rt-thread-standard/api-released-log/kernel-api-released-log

1078249029, 2025-02-16:
`git diff <old_tag> <new_tag> -- 'src/**/*.c'  --function-context`
然后人工最后再判断，这样可以么？

unicornx, 2025-02-17:
> `git diff <old_tag> <new_tag> -- 'src/**/*.c' --function-context` 然后人工最后再判断，这样可以么？

“--function-context" 的用途可能不是我们想要的效果，具体看 <https://yjlv.blogspot.com/2015/06/show-c-functions-in-githg-diffs.html>

> 
> ---update---
> 
> 如果要通过doxygen维护kernel的api信息，可能会涉及src目录么？目前该目录下注释只有重点函数有doxygen注释。这样的话也就可能需要重新组织、补充注释

我初步的想法是：

这个需求本质是希望能快速，全面地抓取所有 API 的改动。

RTT kernel 的 API 应该不仅仅局限于 src 目录吧，我理解目前 RTT 整个仓库中涉及的模块都有可能是我们关注的对象。

可能需要定下规则，确保我们关心的所有 API 都定义或者声明在头文件中（API 包括了结构体，常量，宏，函数，除了函数是声明外，其他应该都是定义）。然后我们可以开发一个脚本 diff 这些头文件集合即可。

这些头文件可以在整理 API 文档过程中整理出来。所以我将整个 issue 也归入了 doxygen 文档整理工作的范畴，但开发工作其实并不直接涉及 doxygen 文档部分。

unicornx, 2025-09-02:
看上去也没有人关心这个事情了，先 close

closed 2025-09-02 by unicornx