RT-Thread/rt-thread #10003: [Bug]  components/drivers/ 中的 examples 和 documentation 中的 example 重复了

unicornx, 2025-02-19:
### RT-Thread Version

master

### Hardware Type/Architectures

N/A

### Develop Toolchain

Other

### Describe the bug

举个例子：

`components/drivers/include/drivers/dev_i2c.h` 中 用 “@code” 和  “@endcode” 括起来的例子和  `documentation/device/i2c/i2c.md` 中 “# I2C Bus Device Usage Example” 重复了，只要保留一份即可。

建议修改方案：

采用 “@code” 在 components/drivers/ 下搜索可以看到这些例子。

对于重复的例子，保留 markdown 中的，删掉 header 文件中的。

即使没有重复，也建议将代码例子全部写到 documentation 下的 markdown 文件中去。

1078249029, 2025-02-19:
assign to me pls

supperthomas, 2025-02-20:
这个先不删了吧。先整理其他的。有时候直接看头文件找代码会比找文档方便一些。文档找起来有时候找不到。

unicornx, 2025-02-20:
> 这个先不删了吧。先整理其他的。有时候直接看头文件找代码会比找文档方便一些。文档找起来有时候找不到。

为啥要保留重复冗余的内容，仅仅是为了程序员看代码方便？直接看 html 或者 markdown 不好么？
另外，我希望文档也是代码的一部分，毕竟我们写 documentation 下的内容也是希望大家去看的，否则维护文档也失去意义了。

@supperthomas 我看你又 reopen 了这个 issue，是否继续？

supperthomas, 2025-02-21:
> > 这个先不删了吧。先整理其他的。有时候直接看头文件找代码会比找文档方便一些。文档找起来有时候找不到。
> 
> 为啥要保留重复冗余的内容，仅仅是为了程序员看代码方便？直接看 html 或者 markdown 不好么？
> 另外，我希望文档也是代码的一部分，毕竟我们写 documentation 下的内容也是希望大家去看的，否则维护文档也失去意义了。
> 
> @supperthomas 我看你又 reopen 了这个 issue，是否继续？
> 

汪老师做的已经很多了，可以先看看其他地方有没有需要优化的地方吧。先close，其他已经有很多issue了慢慢来，也可以问问经常用的人，看看哪些还有做的不是很完善的地方。如何让大家用起来更好用方便。多不了多少冗余的。

closed 2025-02-21 by unicornx

supperthomas, 2025-02-21:
[image]

[image]

汪老师，请问这些结构感觉怪怪的，这个能不能帮忙优化优化？

[image]

unicornx, 2025-02-21:
这些应该是 doxygen 自动生成的 treeview 的顺序吧，这个左边 treeview 的展示顺序和右边对应的页面里的展示顺序是一致的。譬如你点击左边的 “Inter-Thread Communication” 后左边 treeview 出现的子项的顺序就是右边 “Inter-Thread Communication” 页面中内容的展示顺序。

目前还不清楚怎么优化，而且要优化成什么样子呢？本身这个列表就是 API 的项。