RT-Thread/rt-thread #10012: [Bug] hpm的bsp在dist时似乎不会拷贝libraries文件夹，编译过不去

kaidegit, 2025-02-20:
### RT-Thread Version

master

### Hardware Type/Architectures

hpm5301evklite

### Develop Toolchain

GCC

### Describe the bug

```shell
(.venv) D:\Github_D\rt-thread\bsp\hpmicro\hpm5301evklite>scons --dist
scons: Reading SConscript files ...
make distribution....
=> hpm5301evklite
=> components (selective copy)
=> copying components Kconfig files
    => copying Kconfig from components
...
=> copying essential paths
    => copying components/legacy/ipc
...
Note: Skipping system feature: overflow_check
    => copying pin from components\drivers\pin
Note: Skipping system feature: messagequeue
Note: Skipping system feature: small_mem_as_heap
Note: Skipping system feature: event
Note: Skipping system feature: mutex
Note: Skipping system feature: user_main
    => copying serial_v2 from components\drivers\serial
Note: Skipping system feature: heap
Note: Skipping system feature: device
Note: Skipping system feature: legacy
Note: Skipping system feature: timer_soft
Note: Skipping system feature: semaphore
Note: Skipping system feature: idle_hook
Note: Skipping system feature: mempool
Note: Skipping system feature: small_mem
    => copying finsh from components\finsh
Note: Skipping system feature: hook
    => copying device_ipc from components\drivers\ipc
Note: Skipping system feature: components_init
Note: Skipping system feature: console
    => copying msh from components\finsh
    => copying serial from components\drivers\serial
Note: Skipping system feature: mailbox
=> include
=> libcpu
=> src
=> tools
Update configuration files...
=> Generated distribution documentation: D:\Github_D\rt-thread\bsp\hpmicro\hpm5301evklite\dist\project\dist_readme.md
suggest to use command scons --dist [--target=xxx] [--project-name="xxx"] [--project-path="xxx"]
dist project successfully!
Error: the toolchain path (/opt/riscv-gnu-gcc/bin) is not exist, please check 'EXEC_PATH' in path or rtconfig.py.

(.venv) D:\Github_D\rt-thread\bsp\hpmicro\hpm5301evklite>cd dist\project\

(.venv) D:\Github_D\rt-thread\bsp\hpmicro\hpm5301evklite\dist\project>scons --menuconfig
scons: Reading SConscript files ...
warning: PKG_USING_RGPOWER (defined at C:\Users\yekai\Documents\DevTools\env-windows\tools\bin\..\..\packages/packages/peripherals/RgPower/Kconfig:2) has leading or trailing whitespace in its prompt
menuconfig: Kconfig:11: 'libraries/Kconfig' not found (in 'rsource "libraries/Kconfig"'). Check that environment variables are set correctly (e.g. $srctree, which is unset or blank). Also note that unset environment variables expand to the empty string.

(.venv) D:\Github_D\rt-thread\bsp\hpmicro\hpm5301evklite\dist\project>scons -j16
scons: Reading SConscript files ...

scons: *** missing SConscript file 'D:\\Github_D\\rt-thread\\bsp\\hpmicro\\hpm5301evklite\\dist\\libraries\\hpm_sdk\\soc\\HPM5300\\HPM5301\\SConscript'
File "D:\Github_D\rt-thread\bsp\hpmicro\hpm5301evklite\dist\project\SConstruct", line 66, in <module>
```

[image]

### Other additional context

听说hpm的bsp有hpm的人维护，能不能帮忙修修👀

@helloeagleyang

helloeagleyang, 2025-02-21:
> https://github.com/RT-Thread/rt-thread/blob/master/bsp%2Fnrf5x%2Ftools%2Fsdk_dist.py
> 
> 参考这个pr一下，很简单

感谢， https://github.com/RT-Thread/rt-thread/pull/10013 PR 已经提交

helloeagleyang, 2025-02-21:
> ### RT-Thread Version
> master
> 
> ### Hardware Type/Architectures
> hpm5301evklite
> 
> ### Develop Toolchain
> GCC
> 
> ### Describe the bug
> (.venv) D:\Github_D\rt-thread\bsp\hpmicro\hpm5301evklite>scons --dist
> scons: Reading SConscript files ...
> make distribution....
> => hpm5301evklite
> => components (selective copy)
> => copying components Kconfig files
>     => copying Kconfig from components
> ...
> => copying essential paths
>     => copying components/legacy/ipc
> ...
> Note: Skipping system feature: overflow_check
>     => copying pin from components\drivers\pin
> Note: Skipping system feature: messagequeue
> Note: Skipping system feature: small_mem_as_heap
> Note: Skipping system feature: event
> Note: Skipping system feature: mutex
> Note: Skipping system feature: user_main
>     => copying serial_v2 from components\drivers\serial
> Note: Skipping system feature: heap
> Note: Skipping system feature: device
> Note: Skipping system feature: legacy
> Note: Skipping system feature: timer_soft
> Note: Skipping system feature: semaphore
> Note: Skipping system feature: idle_hook
> Note: Skipping system feature: mempool
> Note: Skipping system feature: small_mem
>     => copying finsh from components\finsh
> Note: Skipping system feature: hook
>     => copying device_ipc from components\drivers\ipc
> Note: Skipping system feature: components_init
> Note: Skipping system feature: console
>     => copying msh from components\finsh
>     => copying serial from components\drivers\serial
> Note: Skipping system feature: mailbox
> => include
> => libcpu
> => src
> => tools
> Update configuration files...
> => Generated distribution documentation: D:\Github_D\rt-thread\bsp\hpmicro\hpm5301evklite\dist\project\dist_readme.md
> suggest to use command scons --dist [--target=xxx] [--project-name="xxx"] [--project-path="xxx"]
> dist project successfully!
> Error: the toolchain path (/opt/riscv-gnu-gcc/bin) is not exist, please check 'EXEC_PATH' in path or rtconfig.py.
> 
> (.venv) D:\Github_D\rt-thread\bsp\hpmicro\hpm5301evklite>cd dist\project\
> 
> (.venv) D:\Github_D\rt-thread\bsp\hpmicro\hpm5301evklite\dist\project>scons --menuconfig
> scons: Reading SConscript files ...
> warning: PKG_USING_RGPOWER (defined at C:\Users\yekai\Documents\DevTools\env-windows\tools\bin\..\..\packages/packages/peripherals/RgPower/Kconfig:2) has leading or trailing whitespace in its prompt
> menuconfig: Kconfig:11: 'libraries/Kconfig' not found (in 'rsource "libraries/Kconfig"'). Check that environment variables are set correctly (e.g. $srctree, which is unset or blank). Also note that unset environment variables expand to the empty string.
> 
> (.venv) D:\Github_D\rt-thread\bsp\hpmicro\hpm5301evklite\dist\project>scons -j16
> scons: Reading SConscript files ...
> 
> scons: *** missing SConscript file 'D:\\Github_D\\rt-thread\\bsp\\hpmicro\\hpm5301evklite\\dist\\libraries\\hpm_sdk\\soc\\HPM5300\\HPM5301\\SConscript'
> File "D:\Github_D\rt-thread\bsp\hpmicro\hpm5301evklite\dist\project\SConstruct", line 66, in <module>
> [image]
> 
> ### Other additional context
> 听说hpm的bsp有hpm的人维护，能不能帮忙修修👀
> 
> [@helloeagleyang](https://github.com/helloeagleyang)

感谢反馈，PR已经提交

closed 2025-02-21 by kaidegit