RT-Thread/rt-thread #10028: [Bug] 最新的 riscv musl 工具链编译生成的 helloworld 无法工作

unicornx, 2025-02-24:
### RT-Thread Version

latest

### Hardware Type/Architectures

bsp/cvitek

### Develop Toolchain

Other

### Describe the bug

参考 https://github.com/RT-Thread/rt-thread/issues/9812#issuecomment-2664977960，

用 https://download.rt-thread.org/download/rt-smart/toolchains/riscv64gc-linux-musleabi_for_x86_64-pc-linux-gnu_latest.tar.bz2 这个工具链做了一个 helloworld 运行后报错

```c
#include "stdio.h"

void main()
{
	printf("Hello world!\n");
}
```

静态链接：
```shell
~/ws/bin/rtt-tools-smart/riscv64gc-linux-musleabi_for_x86_64-pc-linux-gnu/bin/riscv64-unknown-linux-musl-gcc -static ./hello.c -o ./hello
```

用 bsp/cvitek 下的

启动后运行 hello，报错如下：
```shell
/ # ./hello 
[I/mm.fault] rt_aspace_fault_try_fix: varea not found at 0x0
[E/libcpu.trap] [FATAL ERROR] Exception 12:Instruction Page Fault

[E/libcpu.trap] scause:0x000000000000000c,stval:(nil),sepc:(nil)

--------------Dump Registers-----------------
Function Registers:
        ra(x1) = 0xffffffc000222710     user_sp = 0x0000000300000000
        gp(x3) = 0xffffffc0002d9760     tp(x4) = 0x0000000000107728
Temporary Registers:
        t0(x5) = 0x0000000000000020     t1(x6) = 0xffffffc00415f158
        t2(x7) = 0x000000000000003a
        t3(x28) = 0xffffffc00415f150    t4(x29) = 0x000000000000001f
        t5(x30) = 0x0000000000000002    t6(x31) = 0x0000000000000002
Saved Registers:
        s0/fp(x8) = 0xffffffc00047bdc0  s1(x9) = 0x00000003000023f0
        s2(x18) = 0x0000000300002400    s3(x19) = 0x00000003000023f0
        s4(x20) = 0xffffffffffffffff    s5(x21) = 0x00000003000023f0
        s6(x22) = 0x00000003000023d0    s7(x23) = 0x00000000000feab2
        s8(x24) = 0x00000000000dcda8    s9(x25) = 0x0000000000105aa8
        s10(x26) = 0x0000000000000008   s11(x27) = 0x0000000000000001
Function Arguments Registers:
        a0(x10) = 0x0000000300000000    a1(x11) = (nil)
        a2(x12) = 0x0000000300000000    a3(x13) = 0xffffffc00047c000
        a4(x14) = 0xffffffc000478000    a5(x15) = 0x0000000000000003
        a6(x16) = 0x0000000000000018    a7(x17) = (nil)
sstatus = 0x0000000200044020
        Supervisor Interrupt Disabled
        Last Time Supervisor Interrupt Enabled
        Last Privilege is User Mode
        Permit to Access User Page
        Not Permit to Read Executable-only Page
satp = 0x800000000008414a
        Current Page Table(Physical) = 0x000000008414a000
        Current ASID = (nil)
        Mode = Page-based 39-bit Virtual Addressing Mode
-----------------Dump OK---------------------
fp = 0xffffffc00047bdc0
please use: addr2line -e ./hello -a -f
 0x0 0xffffffc00024f9c2 0xffffffc0002002c8 0xffffffc00024f9c2 0xffffffc0002002c8 0xffffffc00024f9c2 0xffffffc0002002c8 0xffffffc00024f9c2
[E/libcpu.trap] User Fault, killing thread: hello
```

heyuanjie87, 2025-02-24:
这个最新版可能不是真的最新版

BernardXiong, 2025-02-26:
这个后面固定一个版本吧，也就是说musl gcc正式发一个版本出来

unicornx, 2025-02-26:
> 这个后面固定一个版本吧，也就是说musl gcc正式发一个版本出来

怎么说呢，这个问题我去年 12 月底就报了，到现在也没有看到解决的希望。see https://github.com/RT-Thread/rt-thread/issues/9812#issuecomment-2565472290

unicornx, 2025-03-06:
请问谁是负责出 toolchain 版本的，是否可以关注和认领一下这个问题？

rcitach, 2025-03-10:
上述得到的工具链并不是最新版本，反而是一个非常旧的版本：   
[image]  
在早期的工具链的编译中需要指定连接脚本，但是在userapps的[3ac7f7621d521f27bca4dd44ebb3453157e03f03](https://github.com/RT-Thread/userapps/commit/3ac7f7621d521f27bca4dd44ebb3453157e03f03)删去了,所以使用该工具链默认编译出的应用的入口地址是有问题的  
[image]

可以使用https://github.com/RT-Thread/userapps/blob/main/repo/packages/r/riscv64gc-unknown-smart-musl/xmake.lua 提供的地址，虽然不是最新版本的，但是可以满足当前的开发需求

[image]

[image]

unicornx, 2025-03-12:
感觉这个问题又回到了原点，各种渠道会告诉我有个可以用的版本。

可是我想再说一次的是：我希望 RTT 发布一个正式的可用版本，就像熊大在  <https://github.com/RT-Thread/rt-thread/issues/9812#issuecomment-2664977960> 上说的那种。这些链接拥有固定的地址，这样社区的所有人都可以方便地下载。而且 ***这个地址最好公布到 RTT 的官网上***。

guttatus, 2025-03-24:
如#9812 所述，目前在`bsp/qemu-virt64-riscv`文档中给出的riscv64-unknown-linux-musl-gcc下载链接任存在上述问题，希望能尽快提供可用版本链接