RT-Thread/rt-thread #10032: [Bug] 不知道为什么在打包156工程的时候跳过了很多文件

CYFS3, 2025-02-24:
### RT-Thread Version

master

### Hardware Type/Architectures

nxp

### Develop Toolchain

Other

### Describe the bug

[image]
如图跳过了很多文件，使用scons --dist

kaidegit, 2025-02-25:
https://github.com/RT-Thread/rt-thread/pull/9885

supperthomas, 2025-02-25:
> [#9885](https://github.com/RT-Thread/rt-thread/pull/9885)

看起来按需添加了，可以基于bsp直接开发。
https://club.rt-thread.org/ask/article/21c188cfc2c3df58.html

CYFS3, 2025-02-25:
> [#9885](https://github.com/RT-Thread/rt-thread/pull/9885)

确实感觉有点不合理，如果想打包出来一个文件还得全选完之后才dist，但是我这个已经使能i2c了，打包出来也没有把rt的i2c驱动打包出来

supperthomas, 2025-02-25:
> > 从log看，已经copy了
> > [image]
> 
> 但是并没有拷贝

那就需要检查一下了，可能有bug

Rbb666, 2025-02-28:
如果还没完善好的话感觉可以先把之前revert

supperthomas, 2025-02-28:
我试了下，好像可以成功，但有时候会出现丢文件的情况，可以多试试。action tool里面可以多加一些bsp测试测试。后面可以慢慢完善完善。不用revert掉。

kaidegit, 2025-02-28:
反正我觉得开发者倒是一般不差那几个文件的空间。。。

ZhaoCake, 2025-02-28:
我周末进行修改，考虑获取compile_commands来确保打包文件完整。

supperthomas, 2025-02-28:
> 反正我觉得开发者倒是一般不差那几个文件的空间。。。

欢迎直接在标准版仓库开发。

kaidegit, 2025-03-01:
> 我周末进行修改，考虑获取compile_commands来确保打包文件完整。

可以靠参数调整拷贝全量还是所需嘛☝️

supperthomas, 2025-03-01:
https://github.com/RT-Thread/rt-thread/blob/master/.github/workflows/action_tools.yml
可以多加个bsp，比如STM32L4， nxp等，多验证几种scons --dist的重见用法

closed 2025-03-11 by CYFS3