RT-Thread/rt-thread #10036: [Feature] [doxygen] Should doxygen comment blank lines be removed in params of functions and group attributes?

1078249029, 2025-02-25:
### Describe problem solved by the proposed feature

doxygen注释模板包含空行，但是实践下来去除空可以能获得更好的视觉效果，这在参数比较多的函数中尤其明显
```c
/**
 * @brief Read audio pipe
 * 
 * @param[in] dev pointer to audio device will be read 
 * 
 * @param[in] pos useless param
 * 
 * @param[in] buffer pointer to ringbuffer of audio pipe to be read 
 * 
 * @param[in] size number of bytes will be read
 * 
 * @return error code
 * 
 * @note This function will execute time-consuming or affecting the 
 *             system operations like memcpy and disable interrupt.      
 */
static rt_ssize_t rt_audio_pipe_read(rt_device_t dev,
                              rt_off_t    pos,
                              void       *buffer,
                              rt_size_t   size)
{
    ...
}
```
尤其是位于长文件，拥有多参数的函数中阅读起来比较麻烦。改成下面样式阅读会更方便？
```c
/**
  * @brief Read audio pipe
  * 
  * @param[in] dev pointer to audio device will be read 
  * @param[in] pos useless param
  * @param[in] buffer pointer to ringbuffer of audio pipe to be read 
  * @param[in] size number of bytes will be read
  * @return  error code
  * 
  * @note This function will execute time-consuming or affecting the 
  *              system operations like memcpy and disable interrupt. 
  */
static rt_ssize_t rt_audio_pipe_read(rt_device_t dev,
                              rt_off_t    pos,
                              void       *buffer,
                              rt_size_t   size)
{
    ...
}
```

1078249029, 2025-02-25:
另外在规划doxygen的group结构时是否也需要如此？比如我需要生成Driver->Audio->Audio Device结构来与Driver->Audio->Audio Pipe区分，下面写法会很啰嗦
```c
/**
 * @addtogroup group_Drivers RT-Thread Drivers
 * 
 * @defgroup group_Audio Audio
 * 
 * @brief Audio driver API.
 */

/**
 * @addtogroup group_Audio
 * @{
 */

/**
 * @defgroup group_AudioDevice Audio Device
 * 
 * @brief Audio device API.
 */

/**
 * @addtogroup group_AudioDevice
 * @{
 */

do something

/** @} group_Audio */

/** @} group_Drivers */
```
改成下面的会舒服很多
```c
/**
 * @addtogroup group_Drivers RT-Thread Drivers
 * @defgroup group_Audio Audio
 * @brief Audio driver API.
 * @addtogroup group_Audio
 * @{
 */

/**
 * @defgroup group_AudioDevice Audio Device
 * @brief Audio device API.
 * @addtogroup group_AudioDevice
 * @{
 */

do something

/** @} group_Audio */

/** @} group_Drivers */
```

1078249029, 2025-02-25:
@unicornx

unicornx, 2025-02-25:
保持空行的目的是为了避免把文字全部挤在一起，文字越密集越容易视力疲劳，这对长时间看电脑的程序员来说长期来看是有好处的。

supperthomas, 2025-02-26:
最后生成的html有差异吗？截图看看

unicornx, 2025-02-26:
> 最后生成的html有差异吗？截图看看

of course 没有差异 :)

supperthomas, 2025-02-26:
我试了下，vscode的插件是不含有空格的。

[image]

[image]

百度工具自动注释也是没有空格的。这个应该有空格没有空格都行，注释嘛。
应该让大家写comment越方便和越简单更好，所以建议不用空格，花更多更关心内容如何写好即可。

supperthomas, 2025-02-26:
我试了下，大部分工具和

> 保持空行的目的是为了避免把文字全部挤在一起，文字越密集越容易视力疲劳，这对长时间看电脑的程序员来说长期来看是有好处的。
> 
> 而且不仅空行，在一行里的文字中很多地方我们还希望加空格，具体见 https://worktile.com/kb/ask/2705319.html

这个看起来讲的大部分可能是代码部分。
brief 和param之间还是需要空行的。
param和param之间建议不用空行

supperthomas, 2025-02-26:
@1078249029 你可以参考一下

unicornx, 2025-02-27:
这个问题就不要纠结了吧，我在 doxygen 的指导中好像没有强调必须要加空行吧（如果的确写了 must 啥的欢迎指出我去改成 suggest），给出的所谓模板也就是例子而已。

[image]

所以大家可以多关注注释文档的内容就好。

That's all.

closed 2025-02-28 by 1078249029