RT-Thread/rt-thread #10065: [Bug] audio 的 API doxygen 注释需要改进

unicornx, 2025-03-03:
### RT-Thread Version

master

### Hardware Type/Architectures

N/A

### Develop Toolchain

Other

### Describe the bug

1. `components/drivers/include/drivers/dev_audio.h` 中的 Data Structures 的成员都缺乏注释
2. enum 部分代码上应该给 enum 定义一个名字为好，否则在 html 上只显示一个 enum （doxygen 认为其是一个 anonymous enum），不过这个涉及代码的改动，请检查加上是否会有副作用。
3. rt_audio_ops 缺失了。作为一个需要 user 定义的 callback 函数的集合，需要对回调函数的行为做简单的描述，这对 bsp 使用还是很重要的。
4. 一个改进建议，对于 marco 我看你为了分组新建了许多 group，这个没有问题。只是在对组的 brief 描述上我建议改进一下。譬如对 audio_dsp，注释是 ”Support Dsp(input/output) Units controls“ 感觉说的不是很清楚，这些 marco 是不是每个对应一种类型（type）? 这个在注释上感觉不出来。类似的还有 audio_mixer 和 audio_codec_cmd。

unicornx, 2025-03-03:
@1078249029 请检查，如果接受请认领这个 issue。

1078249029, 2025-03-03:
ok

unicornx, 2025-03-04:
> ok，这些改好后还是提交RFC么？

RFC 的意思是当改动比较大，希望提出来给社区作为一种草案讨论的意思，也就是说当你提交的 pr 不希望合并而只是给大家参考看看的时候的一种说法，引用 https://kernelnewbies.org/PatchTipsAndTricks

> Another common Subject prefix is "RFC". This stands for "Request for Comment", which tells maintainers should review your patch thoroughly, and provide feedback. RFC is typically used when sending feature patches for the first time, or anytime the patch is more than just a simple bug fix.

unicornx, 2025-03-06:
#10073 解决了所列问题的 1/2/4/5/6
还有 3 和 7 有待解决。

1078249029, 2025-03-09:
3 已经在 #10094 中解决了，bsp编译没有问题，还有什么需要注意的么？7 被汇总在 #10095 等待解决

unicornx, 2025-03-12:
3 最终在 #10101 中解决。7 归入 #10095,  本 issue 可以关闭了。

closed 2025-03-12 by unicornx