RT-Thread/rt-thread #10098: [Feature] 请问在RT-Smart中，有没有对于LittleFs这种故障安全的文件系统支持：

WangShuoran, 2025-03-10:
### Describe problem solved by the proposed feature

https://www.rt-thread.org/document/api/group___d_f_s.html
在这里我看到只有FAT32的支持，而且我综合K230平台你们的RT-Smart验证，也是只有FAT32这个文件系统：
[image]
希望能对LittleFs支持，谢谢！

PS：希望更新下当前RT-Smart的文档，与RT-Thread标准版的关系，谢谢！

BernardXiong, 2025-03-19:
littefs应该不行吧，这个文件系统支持不到mmap，page cache。需要支持这些，都需要对接到dfs v2.0

WangShuoran, 2025-03-19:
那有没有像ext2/3/4的这种支持呢？因为用咱们的系统的fat32话，再用sqlite这种文件系统，它的写放大很厉害。

[image]

我有看到lwext4的项目，但是有信息得到在rt thread平台上移植，可能会失败。
请问一下，如果能移植ext 4的话，需要注意哪些事情呢？能给一些参考或者直接支持这种文件系统吗？

BernardXiong, 2025-03-24:
可以使用lwext4，有针对dfs v2.0的版本的。

WangShuoran, 2025-04-05:
能不能指路具体支持dfs v2.0的位置？