RT-Thread/rt-thread #10154: [Bug] bsp/cvitek AARCH：启动内核时控制台打印乱码

unicornx, 2025-03-28:
### RT-Thread Version

master

### Hardware Type/Architectures

bsp/cvitek

### Develop Toolchain

Other

### Describe the bug

发现在 AARCH64 大核启动过程中，a77a86d6b2 (tag: GOOD_20250204) 时还是正常的，但是切换到  74f43edd6c (tag: GOOD_20250211) 时内核启动后会打印乱码，然后 logo 时恢复正常。
这个问题在 RISC-V 大核上没有看到。

```
74f43edd6c (HEAD, tag: GOOD_20250211) [DM/THERMAL] Remove unused code in update poll
aaf18e497c [DM/LED] Rename LED register/unregister
f506076281 [DM] Replace spinlock static init by RT_DEFINE_SPINLOCK
f20e8973be [DFS/ISO9660] Remove warning for ops
5a455cb615 [DM/PIN] Reset the value of PIN_NONE
578cfc0b2a [DM/CORE] Add IDA init in runtime
d41a0351db [AARCH64] Update kernel's boot link for ARM64
1564735b5c [DM/SPI] Make CS pin config fixed in system (#9977)
f820c1948a [drivers/spi]修复spi configure会执行两次的问题 (#9972)
21ea7c675c 更新 bsp_buildings.yml
4edfa4895c ci: remove spell_check
12fded19f5 [DM/FDT] Fixup earlycon loss old messages.
d3d33ff983 [dfs]mmap的文件在关闭后释放file指针 (#9917)
d83d71cc05 [drivers/spi]统一软件SPI配置
6e3c9acd39 [lwp]修正kill(pid < -1)时的返回值错误
a63b97e316 [bsp][nxp]support FRDM-MCXA156 (#9971)
8629c9572a [action]add online packages checking in rt-spark. (#9969)
a77a86d6b2 (tag: GOOD_20250204) bsp: cvitek: fix build error for aarch64
```

出问题的控制台截图， 正常应该没有这行打印：

[image]

具体是 a77a86d6b2 ~ 74f43edd6c 之间哪个 patch 引入的，还需要继续定位。

unicornx, 2025-03-28:
[image]

ziyu04, 2025-03-28:
[image]

[image]
这个打印出来的信息，可能因编码方式不一样，接收到的信息不太一样。
我按照GUI老师的意见翻一下aarch64的源码吧。

1078249029, 2025-03-28:
ok，这周末我测一下

1078249029, 2025-03-30:
最新的主线上start kernel之前并没有乱码，我这里出问题的是bootrom的阶段
```shell
B▒▒▒▒▒0/▒▒▒D.▒▒▒L.S▒▒/2▒000▒0▒/▒▒▒▒▒▒0.B▒▒SD.P▒.▒D/0▒0/0x▒▒00/0x100▒▒0▒▒E.▒S.▒D▒▒▒▒000▒▒▒▒▒▒▒▒0xc200/▒▒▒E▒J▒
FSBL zIXgg9:gbeb1483:2024-04-20T23:53:12+08:00
st_on_reason=d0000
st_off_reason=0
P2S/0x1000/0x4010e200.
SD/0xd200/0x1000/0x1000/0.P2E.
DPS/0xe200/0x2000.
SD/0xe200/0x2000/0x2000/0.DPE.
DDR init.
ddr_param[0]=0x78075562.
pkg_type=5
D1_3_2
DDR3-2G-QFN
Data rate=1866.
DDR BIST PASS
PLLS.
PLLE.
C2S/0x10200/0x8fe00000/0x12200.
SD/0x10200/0x12200/0x12200/0.RSC.
C2E.
MS/0x22400/0x80000000/0x7200.
SD/0x22400/0x7200/0x7200/0.ME.
L2/0x29600.
SD/0x29600/0x200/0x200/0.L2/0x414d3342/0xcafef5b9/0x80200000/0x2fc00/0x2fc00
COMP/1.
SD/0x29600/0x2fc00/0x2fc00/0.DCP/0x80200020/0x1000000/0x81900020/0x2fc00/1.
DCP/0x79868/0.
Loader_2nd loaded.
Use internal 32k
Jump to monitor at 0x80000000.
J1651545748.ERROR:   Error initializing runtime service opteed_fast

U-Boot 2021.10-ga57aa1f29b (Apr 20 2024 - 23:53:08 +0800)cvitek_cv181x

DRAM:  254 MiB
MMC:   cv-sd@4310000: 0
Loading Environment from nowhere... OK
In:    serial
Out:   serial
Err:   serial
Net:
Warning: ethernet@4070000 (eth0) using random MAC address - de:67:79:e9:b6:37
eth0: ethernet@4070000
Hit any key to stop autoboot:  0
Boot from SD dev 0 ...
switch to partitions #0, OK
mmc0 is current device
176548 bytes read in 39 ms (4.3 MiB/s)
## Loading kernel from FIT Image at 81800000 ...
   Using 'config-sg2002_milkv_duo256m_arm_sd' configuration
   Trying 'kernel-1' kernel subimage
     Description:  cvitek kernel
     Type:         Kernel Image
     Compression:  lzma compressed
     Data Start:   0x818000d8
     Data Size:    154600 Bytes = 151 KiB
     Architecture: AArch64
     OS:           Linux
     Load Address: 0x80200000
     Entry Point:  0x80200000
     Hash algo:    crc32
     Hash value:   93d93efb
   Verifying Hash Integrity ... crc32+ OK
## Loading fdt from FIT Image at 81800000 ...
   Using 'config-sg2002_milkv_duo256m_arm_sd' configuration
   Trying 'fdt-sg2002_milkv_duo256m_arm_sd' fdt subimage
     Description:  cvitek device tree - sg2002_milkv_duo256m_arm_sd
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x81825de0
     Data Size:    20046 Bytes = 19.6 KiB
     Architecture: AArch64
     Hash algo:    sha256
     Hash value:   c9808286cbf8cafd86d219df4eb848db88b181fab6b5a595bcb088889433d69c
   Verifying Hash Integrity ... sha256+ OK
   Booting using the fdt blob at 0x81825de0
   Uncompressing Kernel Image
   Decompressing 606928 bytes used 51ms
   Loading Device Tree to 000000008d5f1000, end 000000008d5f8e4d ... OK

Starting kernel ...

[I/rtdm.ofw] Booting RT-Thread on physical CPU 0x0
[I/rtdm.ofw] Machine model: SOPHGO ASIC. ARM.
[I/rtdm.ofw] Memory node(1) ranges: 0x0000000080000000 - 0x000000008fe00000
[E/rtdm.ofw] Allocating reserved memory in setup is not yet supported
[E/rtdm.ofw] Allocating reserved memory in setup is not yet supported
[I/mm.memblock] System memory:
[I/mm.memblock]                    [0x0000000080000000, 0x000000008fe00000]
[I/mm.memblock] Reserved memory:
[I/mm.memblock]                    [0x0000000080000000, 0x0000000080080000]
[I/mm.memblock]                    [0x0000000080200000, 0x00000000802df000]
[I/mm.memblock]                    [0x00000000802df000, 0x00000000842df000]
[I/mm.memblock]                    [0x00000000842df000, 0x00000000844df000]
[I/mm.memblock]                    [0x00000000844df000, 0x00000000844e4000]
[I/mm.memblock] physical memory region [0x0000000080080000-0x0000000080200000] installed to system page
[I/mm.memblock] physical memory region [0x00000000844e4000-0x000000008fe00000] installed to system page
[I/mm.memblock] 195 MB memory installed to system page
[I/drv.pinmux] Pin Name = "UART0_RX", Func Type = 281, selected Func [0]

[I/drv.pinmux] Pin Name = "UART0_TX", Func Type = 282, selected Func [0]

[I/drv.pinmux] Pin Name = "IIC0_SDA", Func Type = 285, selected Func [1]

[I/drv.pinmux] Pin Name = "IIC0_SCL", Func Type = 286, selected Func [1]

[I/osi.psci] Using PSCI v1.0 Function IDs
[I/rtdm.ofw] Console: uart0 (<no-node>)

 \ | /
- RT -     Thread Operating System
 / | \     5.2.1 build Mar 29 2025 22:02:55
 2006 - 2024 Copyright by RT-Thread team
[I/rtdm.mnt] File system initialization done
Hello AARCH64 !
msh />

```

unicornx, 2025-03-31:
> 最新的主线上start kernel之前并没有乱码，我这里出问题的是bootrom的阶段

能给出 “最新主线” 对应的 commit ID 么？我再试试。

1078249029, 2025-03-31:
> > 最新的主线上start kernel之前并没有乱码，我这里出问题的是bootrom的阶段
> 
> 能给出 “最新主线” 对应的 commit ID 么？我再试试。

4acef9b1cdde25580765b20f8396a70d98e6b47e

这个commit的arm没什么问题，但riscv核不能启动，终端上都是乱码

unicornx, 2025-03-31:
> 这个commit的arm没什么问题，但riscv核不能启动，终端上都是乱码

我这里现象的确和你不一样，我这里基于 master 的 c97c9c6c57（也就是比 4acef9b1cd 新一个 commit）。看到的现象还是：
- 大核打印 “Starting kernel ...” 后开始乱码输出，直到出现 RTT logo 恢复正常
- 小核打印正常

我用的是 minicom，FT232R 的 usb转 uart ：
```
[ 8163.476860] usb 1-1: new full-speed USB device number 9 using xhci_hcd
[ 8163.609131] usb 1-1: New USB device found, idVendor=0403, idProduct=6001, bcdDevice= 6.00
[ 8163.609150] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 8163.609157] usb 1-1: Product: FT232R USB UART
[ 8163.609162] usb 1-1: Manufacturer: FTDI
[ 8163.609167] usb 1-1: SerialNumber: A50285BI
[ 8163.613761] ftdi_sio 1-1:1.0: FTDI USB Serial Device converter detected
[ 8163.613850] usb 1-1: Detected FT232R
[ 8163.614853] usb 1-1: FTDI USB Serial Device converter now attached to ttyUSB0
```

1078249029, 2025-03-31:
> 我用的是 minicom，FT232R 的 usb转 uart , 下面是 dmesg 的 log：

我这里复现了，可能是minicom的问题？

```shell
U-Boot 2021.10-ga57aa1f29b (Apr 20 2024 - 23:53:08 +0800)cvitek_cv181x

DRAM:  254 MiB
MMC:   cv-sd@4310000: 0
Loading Environment from nowhere... OK
In:    serial
Out:   serial
Err:   serial
Net:
Warning: ethernet@4070000 (eth0) using random MAC address - 6e:2f:02:d5:07:ef
eth0: ethernet@4070000
Hit any key to stop autoboot:  0
Boot from SD dev 0 ...
switch to partitions #0, OK
mmc0 is current device
176436 bytes read in 39 ms (4.3 MiB/s)
## Loading kernel from FIT Image at 81800000 ...
   Using 'config-sg2002_milkv_duo256m_arm_sd' configuration
   Trying 'kernel-1' kernel subimage
     Description:  cvitek kernel
     Type:         Kernel Image
     Compression:  lzma compressed
     Data Start:   0x818000d8
     Data Size:    154488 Bytes = 150.9 KiB
     Architecture: AArch64
     OS:           Linux
     Load Address: 0x80200000
     Entry Point:  0x80200000
     Hash algo:    crc32
     Hash value:   089365c9
   Verifying Hash Integrity ... crc32+ OK
## Loading fdt from FIT Image at 81800000 ...
   Using 'config-sg2002_milkv_duo256m_arm_sd' configuration
   Trying 'fdt-sg2002_milkv_duo256m_arm_sd' fdt subimage
     Description:  cvitek device tree - sg2002_milkv_duo256m_arm_sd
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x81825d70
     Data Size:    20046 Bytes = 19.6 KiB
     Architecture: AArch64
     Hash algo:    sha256
     Hash value:   c9808286cbf8cafd86d219df4eb848db88b181fab6b5a595bcb088889433d69c
   Verifying Hash Integrity ... sha256+ OK
   Booting using the fdt blob at 0x81825d70
   Uncompressing Kernel Image
   Decompressing 604736 bytes used 51ms
   Loading Device Tree to 000000008d5f1000, end 000000008d5f8e4d ... OK

Starting kernel ...

���������������������������������������������������������������������������������������������������������������������������������������������������������������)

 \ | /
- RT -     Thread Operating System
 / | \     5.2.1 build Mar 31 2025 11:24:33
 2006 - 2024 Copyright by RT-Thread team
[I/rtdm.mnt] File system initialization done
Hello AARCH64 !
msh />
msh />
msh />
```

我将串口监视器从mobaxterm变成minicom就复现了，连commit的版本都没改

usb转串口用的是ch340g

```shell
[ 8695.010558] usb 1-1: new full-speed USB device number 2 using vhci_hcd
[ 8695.160494] usb 1-1: SetAddress Request (2) to port 0
[ 8695.204963] usb 1-1: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[ 8695.204970] usb 1-1: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 8695.204972] usb 1-1: Product: USB Serial
[ 8695.244646] usbcore: registered new interface driver ch341
[ 8695.244711] usbserial: USB Serial support registered for ch341-uart
[ 8695.258651] usb 1-1: ch341-uart converter now attached to ttyUSB0
```

之前在用mobaxterm测试的时候也复现了一次问题，但是我重启mobaxterm问题就消失了，不知道跟这个问题有没有关系

1078249029, 2025-04-01:
目前问题已经定位到 12fded19f531a4711e36a19de1f1f5055c1be110 这个分支了

PR #10166 (2025-04-14): [DM/FDT] Fix garble when booting
## 拉取/合并请求描述：(PR description)

[

#### 为什么提交这份PR (why to submit this PR)
Fix garble caused by empty implementation of earlycon series function in driver.If driver doesn't offer earlycon_id or rt_f dt_earlycon_id.setup function but open option, the checking of best_earlycon_id will failed.

Fixed #10154 

#### 你的解决方案是什么 (what is your solution)

#### 请提供验证的bsp和config (provide the config and bsp) 

- BSP:

- .config:
closed 2025-04-14 by Rbb666