RT-Thread/rt-thread #10156: [Feature] 建立POSIX SIG

BernardXiong, 2025-03-28:
### Describe problem solved by the proposed feature

POSIX（Portable Operating System Interface，可移植操作系统接口）是由IEEE制定的跨平台操作系统接口标准，旨在统一UNIX-like系统的应用程序接口（API），确保软件在不同系统间的可移植性。

POSIX是RT-Thread上极为看重的一点，需要在这块上做得更完善，更好。

### Describe your preferred solution

### POSIX标准概述

#### 1. **POSIX的定义与核心目标**
POSIX（Portable Operating System Interface，可移植操作系统接口）是由IEEE制定的跨平台操作系统接口标准，旨在统一UNIX-like系统的应用程序接口（API），确保软件在不同系统间的可移植性。其核心目标包括：
- **标准化接口**：定义文件操作、进程控制、线程管理、信号处理等基础API，形成统一的编程规范。
- **跨平台兼容**：通过规范源代码级接口，使开发者无需修改代码即可在不同系统（如Linux、macOS、嵌入式RTOS）上运行程序。
- **模块化扩展**：POSIX标准分为多个子集（如POSIX.1、POSIX.1b等），支持按需实现特定功能。

#### 2. **POSIX的版本与结构**
- **核心版本**：POSIX.1是最基础的规范，涵盖进程管理、内存分配、错误处理等系统接口。
- **集成标准**：后续版本如POSIX.1b（实时扩展）、POSIX.1c（线程）逐步加入高级功能。
- **统一整合**：2008年后，多个子标准合并为IEEE Std 1003.1-2008（即POSIX.1-2008），包含基础定义、系统接口、Shell工具等模块。

---

### PSE51/52/53/54：实时与嵌入式系统配置文件
这些子标准属于**IEEE 1003.13-2003**，专为实时和嵌入式系统设计，按功能递增分为四层：

#### 1. **PSE51：最小实时系统配置文件（Minimal Real-time System Profile）**
- **适用场景**：资源极度受限的嵌入式设备（如传感器节点）。
- **核心特性**：
  - 仅支持单进程，无多线程或多进程功能。
  - 不包含文件系统或进程间通信（IPC）。
  - 提供基础API如信号处理（`sigaction()`）、定时器（`nanosleep()`）和内存锁定（`mlock()`）等。
- **典型应用**：AP AUTOSAR的应用程序层。

#### 2. **PSE52：实时控制器配置文件（Real-time Controller Profile）**
- **功能扩展**：在PSE51基础上增加多线程、信号量及消息队列。
- **关键接口**：
  - 线程调度（`pthread_create()`）、互斥锁（`pthread_mutex_lock()`）。
  - 实时消息传递（`mq_send()`）和同步I/O（`msync()`）。

#### 3. **PSE53：专用实时集成配置文件（Dedicated Real-time Integration Profile）**
- **增强功能**：
  - 引入共享内存（`shm_open()`）、高级信号跳转（`sigsetjmp()`）和内存保护（`mprotect()`）等。
  - 支持多进程（`fork()`、`vfork()`）和管道通信（`pipe()`）。
- **典型实现**：RT-Thread Smart操作系统用户态兼容PSE53，适用于需要复杂通信的机器人控制。

#### 4. **PSE54：通用实时集成配置文件（General Real-time Integration Profile）**
- **完整特性**：
  - 包含完整文件系统（`chmod()`）、套接字（`socket()`）和用户权限管理（`geteuid()`）。
  - 支持多用户环境和Shell工具（如`fnmatch()`）。

### Describe possible alternatives

成立POSIX SIG，对POSIX进行持续改善。

BernardXiong, 2025-03-28:
_from https://www.cnblogs.com/wsg1100/p/18166587_

**RT-Thread**: 5.x版本虽然实现了hrtimer逻辑，但是你用时会发现是个半成品，可以选择一个硬件Timer来实现hrtimer底层定时，但是这样每次只能有一个任务处于定时状态（用全局变量来维护timer的参数，且ctime层 **hrtime、cputime接口职责隔离不清晰，无可扩展性，可以看出"能跑就行"，质量堪忧**）；默认情况下还是基于Tick。

supperthomas, 2025-04-02:
这里反馈一下，在使用posix的时候，有些代码中的类型很大概率可能索引到toolchain中的头文件，比如目前bsp中的任意bsp都会包含components\libc\compilers\common\ctime.c, 这个函数中clock_t 的类型在rtthread的代码中未找到定义，很大概率有些平台编译不过。

```
rt_weak clock_t clock(void)
{
    return rt_tick_get();  // TODO should return cpu usage time
}
RTM_EXPORT(clock);
```

supperthomas, 2025-04-02:
```
# scons
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build
CC build\applications\main.o
In file included from J:\02_project\01_repo\github\ci_file\rt-thread\components\libc\compilers\common\include/sys/time.h:18,
                 from J:\02_project\01_repo\github\ci_file\rt-thread\components\libc\compilers\common\include/sys/select.h:17,
                 from j:\06_software\01_green_software\xpack-riscv-none-embed-gcc-10.2.0-1.2-win32-x64\xpack-riscv-none-embed-gcc-10.2.0-1.2\riscv-none-embed\include\sys\types.h:50,
                 from j:\06_software\01_green_software\xpack-riscv-none-embed-gcc-10.2.0-1.2-win32-x64\xpack-riscv-none-embed-gcc-10.2.0-1.2\riscv-none-embed\include\stdio.h:61,
                 from applications\main.c:11:
j:\06_software\01_green_software\xpack-riscv-none-embed-gcc-10.2.0-1.2-win32-x64\xpack-riscv-none-embed-gcc-10.2.0-1.2\riscv-none-embed\include\time.h:56:1: error: unknown type name 'clock_t'; did you mean '_flock_t'?
   56 | clock_t    clock (void);
      | ^~~~~~~
      | _flock_t
In file included from J:\02_project\01_repo\github\ci_file\rt-thread\components\libc\compilers\common\include/sys/select.h:17,
                 from j:\06_software\01_green_software\xpack-riscv-none-embed-gcc-10.2.0-1.2-win32-x64\xpack-riscv-none-embed-gcc-10.2.0-1.2\riscv-none-embed\include\sys\types.h:50,
                 from j:\06_software\01_green_software\xpack-riscv-none-embed-gcc-10.2.0-1.2-win32-x64\xpack-riscv-none-embed-gcc-10.2.0-1.2\riscv-none-embed\include\stdio.h:61,
                 from applications\main.c:11:
J:\02_project\01_repo\github\ci_file\rt-thread\components\libc\compilers\common\include/sys/time.h:55:5: error: unknown type name 'suseconds_t'
   55 |     suseconds_t tv_usec;    /* and microseconds */
      |     ^~~~~~~~~~~
scons: *** [build\applications\main.o] Error 1
scons: building terminated because of errors.
```

snikeguo, 2025-05-13:
posix 对软件能力要求极高，尤其是实现sigfpe这种信号，要不然我就不会用nuttx了。。