RT-Thread/rt-thread #10161: [Feature] [risc-v] RISC-V 的toolchain 讨论

supperthomas, 2025-04-01:
### Describe problem solved by the proposed feature

目前RISC-V的toolchain版本比较多，这里加一些讨论，讨论哪些toolchain版本是比较通用和官方的，方便后续RISC-V平台整理和使用比较统一的版本。
大家可以提供一些自己常用的RISCV的toolchain的下载路径和RISCV平台的详细信息。

supperthomas, 2025-04-01:
https://github.com/riscv-collab/riscv-gnu-toolchain/releases

这个是RISCV的版本

supperthomas, 2025-04-01:
https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/releases

这个是XPACK的版本

supperthomas, 2025-04-01:
```yml
riscv-none-embed-gcc:
  bsp:
  - gd32/risc-v/gd32vf103r-start
  - gd32/risc-v/gd32vf103v-eval
  - hifive1
  - juicevm
  - k210
  - rv32m1_vega/ri5cy
  - wch/risc-v/ch32v103r-evt
  - wch/risc-v/ch32v208w-r0
  - wch/risc-v/ch32v307v-r1
  - wch/risc-v/ch569w-evt
  - wch/risc-v/yd-ch32v307vct6
  count: 11
  download_url: https://github.com/RT-Thread/toolchains-ci/releases/download/v1.5/xpack-riscv-none-embed-gcc-8.3.0-2.3-linux-x64.tar.gz
riscv-nuclei-elf-gcc:
  bsp:
  - nuclei/gd32vf103_rvstar
  - nuclei/nuclei_fpga_eval
  count: 2
riscv32-esp-elf-gcc:
  bsp:
  - ESP32_C3
  count: 1
  download_url: https://github.com/espressif/crosstool-NG/releases/download/esp-2022r1-RC1/riscv32-esp-elf-gcc11_2_0-esp-2022r1-RC1-linux-amd64.tar.xz
riscv32-unknown-elf-gcc:
  bsp:
  - core-v-mcu/core-v-cv32e40p
  - hpmicro/hpm5300evk
  - hpmicro/hpm5301evklite
  - hpmicro/hpm6200evk
  - hpmicro/hpm6300evk
  - hpmicro/hpm6750evk
  - hpmicro/hpm6750evk2
  - hpmicro/hpm6750evkmini
  - hpmicro/hpm6800evk
  count: 9
  download_url: https://github.com/hpmicro/riscv-gnu-toolchain/releases/download/2022.05.15/riscv32-unknown-elf-newlib-multilib_2022.05.15_linux.tar.gz
riscv64-unknown-elf-gcc:
  bsp:
  - bluetrum/ab32vg1-ab-prougen
  - bouffalo_lab/bl60x
  - bouffalo_lab/bl61x
  - bouffalo_lab/bl70x
  - bouffalo_lab/bl808/lp
  - bouffalo_lab/bl808/m0
  - cvitek/c906_little
  - cvitek/cv18xx_risc-v
  - qemu-virt64-riscv
  - sparkfun-redv
  - thead-smart
  count: 11
  download_url: https://github.com/RT-Thread/toolchains-ci/releases/download/v1.4/riscv64-unknown-elf-toolchain-10.2.0-2020.12.8-x86_64-linux-ubuntu14.tar.gz
riscv64-unknown-linux-musl-gcc:
  bsp:
  - allwinner/d1
  - allwinner/d1s
  - bouffalo_lab/bl808/d0
  count: 3
sparc-gaisler-elf-gcc:
  bsp:
  - bm3803
  count: 1
unicore32-linux-gcc:
  bsp:
  - sep6200
  count: 1
```

supperthomas, 2025-04-01:
字段含义：
32bit：表示处理器支持 32 位指令集。
a：表示支持 RISC-V 的 A 扩展（原子操作指令集），用于实现原子操作。
c：表示支持 RISC-V 的 C 扩展（压缩指令集），用于减少指令长度。
i：表示支持 RISC-V 的基础指令集 I（Integer Instruction Set），是 RISC-V 的核心指令集。
m：表示支持 RISC-V 的 M 扩展（整数乘除指令集），用于执行整数乘法和除法。
zaamo：表示支持 RISC-V 的 Zaamo 扩展，这是从标准 A 扩展拆分出来的部分，专注于原子操作。
zalrsc：表示支持 RISC-V 的 Zalrsc 扩展，这是从标准 A 扩展拆分出来的部分，专注于 LR/SC（Load-Reserved/Store-Conditional）指令。
zicsr：表示支持 RISC-V 的 Zicsr 扩展，用于访问控制和状态寄存器（CSR）。
zifencei：表示支持 RISC-V 的 Zifencei 扩展，用于指令缓存一致性操作。
zmmul：表示支持 RISC-V 的 Zmmul 扩展，用于实现乘法指令。

supperthomas, 2025-04-01:
[image]

supperthomas, 2025-04-01:
统一来看
32bit的嵌入式非linux平台可以用：
https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/releases

64bit和32bit都可以使用，这个下面链接只有ubuntu版本，没看到windows版本：
https://github.com/riscv-collab/riscv-gnu-toolchain/releases

supperthomas, 2025-04-01:
https://github.com/xpack-dev-tools/riscv-none-embed-gcc-xpack/releases
这个也比较好用，最后版本是10.2.0-1.2 不更新维护了。

supperthomas, 2025-04-01:
> 这个议题是不是没有考虑 RT-smart 的情况？

是的，这块我没有接触。有经验的也可以讨论。给出一些参考。

unicornx, 2025-04-02:
补充一些我这边维护比较多的 riscv 相关 bsp 使用的 toolchain 信息(包括了 RT-standard 和 RT-smart)：

cvitek（milkv-duo 系列）: https://github.com/RT-Thread/rt-thread/blob/master/bsp/cvitek/README.md#41-toolchain-%E4%B8%8B%E8%BD%BD

virt64-riscv: https://github.com/RT-Thread/rt-thread/blob/master/bsp/qemu-virt64-riscv/README_cn.md#21-%E5%AE%89%E8%A3%85%E5%B7%A5%E5%85%B7%E9%93%BE

另外，我一直在跟踪的 riscv 工具链不统一的问题，提的相关 issue:
- https://github.com/RT-Thread/rt-thread/issues/10028
- https://github.com/RT-Thread/rt-thread/issues/9812
感觉推动很困难，拖了很长时间了。

supperthomas, 2025-04-02:
> 补充一些我这边维护比较多的 riscv 相关 bsp 使用的 toolchain 信息(包括了 RT-standard 和 RT-smart)：
> 
> cvitek（milkv-duo 系列）: https://github.com/RT-Thread/rt-thread/blob/master/bsp/cvitek/README.md#41-toolchain-%E4%B8%8B%E8%BD%BD
> 
> virt64-riscv: https://github.com/RT-Thread/rt-thread/blob/master/bsp/qemu-virt64-riscv/README_cn.md#21-%E5%AE%89%E8%A3%85%E5%B7%A5%E5%85%B7%E9%93%BE
> 
> 另外，我一直在跟踪的 riscv 工具链不统一的问题，提的相关 issue:
> 
> * [[Bug] 最新的 riscv musl 工具链编译生成的 helloworld 无法工作 #10028](https://github.com/RT-Thread/rt-thread/issues/10028)
> * [[Question] 有没有固定的下载地址可以得到最新的可用用于开发 riscv 的 musl gcc？感觉一直没有搞清楚 #9812](https://github.com/RT-Thread/rt-thread/issues/9812)
> 
> 感觉推动很困难，拖了很长时间了。

嗯，标准版理论上应该可以用官方的。下面的链接对于RISCV64和RISCV32都是兼容的
https://github.com/xpack-dev-tools/riscv-none-embed-gcc-xpack/releases
或者
https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/releases

supperthomas, 2025-04-02:
SMART 版本可以尝试下下面的链接：
https://github.com/riscv-collab/riscv-gnu-toolchain/releases
选择musl版本

unicornx, 2025-04-02:
> SMART 版本可以尝试下下面的链接： https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2025.01.20/riscv64-musl-ubuntu-22.04-gcc-nightly-2025.01.20-nightly.tar.xz 选择musl版本，这块没经验，不确定，只能建议尝试，后面会出一些文章介绍相关术语

这个估计不行，rt-smart 用的 musl gcc 听说是定制了 rtt 自己的什么东西，具体要问睿赛德了。

unicornx, 2025-04-02:
> 嗯，标准版理论上应该可以用官方的。下面的链接对于RISCV64和RISCV32都是兼容的 https://github.com/xpack-dev-tools/riscv-none-embed-gcc-xpack/releases 或者 https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/releases

但是要注意类似 duo 这些使用的 c906 里的 cpu core 上会用到一些 vendor 自己的扩展，譬如 v-ext v0.7, 这些 gcc 主线不一定支持的，所以说还不能完全确定。

zhangjing0303, 2025-04-02:
> > SMART 版本可以尝试下下面的链接： https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2025.01.20/riscv64-musl-ubuntu-22.04-gcc-nightly-2025.01.20-nightly.tar.xz 选择musl版本，这块没经验，不确定，只能建议尝试，后面会出一些文章介绍相关术语
> 
> 这个估计不行，rt-smart 用的 musl gcc 听说是定制了 rtt 自己的什么东西，具体要问睿赛德了。

应该是的，nusl libc库是定制过的，前几天我刚问过

supperthomas, 2025-04-02:
嗯，首先确定的是：
标准版和smart版不能共用一个toolchain。

supperthomas, 2025-04-02:
标准版就是当前仓库版本倒是可以统一用一个，当然也不是强制的，只是当大家不知道用哪款toolchain的时候，给一个统一的建议。当然也有一些特殊的指令拓展像c906，大家也可以把遇到的不能用的也列出来。

BernardXiong, 2025-04-02:
smart工具链可另看。

riscv的工具链问题确实头大，开源的ISA，然后工具链一堆的发散，收敛不了，很头疼，我在公开场合也提到过……

supperthomas, 2025-04-02:
https://releases.linaro.org/components/toolchain/binaries/

最后2019年12月，已经没有继续更新了

helloeagleyang, 2025-04-15:
HPMicro用的toolchain是基于标准的riscv-gnu-toolchain仓库编译出来的，只包含了RV32系列的，是xpack版本的子集，理论上可以换成xpack版本的。https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/releases/tag/v14.2.0-3 这个似乎是最新的GCC14.x版本的

kaidegit, 2025-04-16:
wch对mrs的toolchain写的编译选项，换成xpack可能会有点问题。。。快速中断和代码压缩啥的。不过目前看来似乎并不影响编译。

可以参考
https://github.com/cjacker/opensource-toolchain-ch32v

kaidegit, 2025-04-16:
hpm的用xpack的问题不大，hslink项目都在用

supperthomas, 2025-04-18:
zephyr 的toolchain，也只用了一种RISCV64的toolchain，兼容RISCV32：

https://github.com/zephyrproject-rtos/sdk-ng/releases/download/v0.17.1-rc1/toolchain_linux-x86_64_riscv64-zephyr-elf.tar.xz

Target | Linux | macOS | Windows
-- | -- | -- | --
riscv64-zephyr-elf | [AArch64][toolchain_gnu_linux-aarch64_riscv64-zephyr-elf.tar.xz] / [x86-64][toolchain_gnu_linux-x86_64_riscv64-zephyr-elf.tar.xz] | [AArch64][toolchain_gnu_macos-aarch64_riscv64-zephyr-elf.tar.xz] / [x86-64][toolchain_gnu_macos-x86_64_riscv64-zephyr-elf.tar.xz] | [x86-64][toolchain_gnu_windows-x86_64_riscv64-zephyr-elf.7z]
-- | -- | -- | --

supperthomas, 2025-04-18:
RISC-V指令集由“基本指令集(I) + 扩展指令集(E)”组成。

基本指令(I)集是必选的，扩展指令集(E)是可选的。意思就是可以根据你的实际需求，选择需要使用的指令。例如在一个项目中，如果不需要用到压缩指令，那么就不需要把压缩指令添加进来，从而做到定制化，这也是RISC-V的一大特点。

RISC-V 指令集有 RV32I、RV32E、RV64I、RV64E、RV64I 等等。

RV 代表 RISC-V，32/64 代表 32 位或 64 位，I 和 E 都是基本指令集，在 I 和 E 的基础上，可以添加 D（双精度浮点扩展）、M（整数乘除法）、A（原子扩展）、C（压缩扩展）等扩展指令。

例如，在 RV64I 基础上，添加原子、整数乘除法、双精度浮点、压缩指令，则该指令集称为 RV64IMADC。

supperthomas, 2025-04-18:
基本指令集
指令集名称	描述	版本	状态
RV32I	基本整数指令集, 32位元	2.1	已批准
RV32E	基本整数指令集(嵌入式系统), 32位元, 16 个暂存器	2.0	已批准
RV64I	基本整数指令集, 64位元	2.1	已批准
RV64E	基本整数指令集(嵌入式系统), 64位元, 16 个暂存器	2.0	已批准
RV128I	基本整数指令集, 128位元	1.7	开放

标准扩充指令集
指令集名称	描述	版本	状态
M	整数乘除法标准扩充	2.0	已批准
A	不可中断指令(Atomic)标准扩充	2.1	已批准
F	单精度浮点标准扩充	2.2	已批准
D	双精度浮点标准扩充	2.2	已批准
Zicsr	控制与状态寄存器	2.0	已批准
Zifencei	指令抓取屏障	2.0	已批准
G	所有以上的扩充指令集以及基本指令集的总和的简称	不适用	不适用
Q	四精度浮点标准扩充	2.2	已批准
L	十进位浮点标准扩充	0.0	开放
C	压缩指令标准扩充	2.0	已批准
B	位元运算标准扩充	1.0	已批准
J	动态指令翻译标准扩充	0.0	开放
T	顺序记忆体存取标准扩充	0.0	开放
P	单指令多资料流（SIMD）运算标准扩充	0.9.10	开放
V	向量运算标准扩充	1.0	冻结
Zk	标量加密标准扩充	1.0.1	已批准
H	Hypervisor 标准扩充	1.0	已批准
S	Supervisor 标准扩充	1.12	已批准
Zam	非对齐不可中断指令标准扩展	0.1	开放
Zihintpause	暂停提示	2.0	已批准
Zihintntl	非时间局部性提示	0.2	开放
Zfa	额外浮点运算指令标准扩展	0.1	开放
Zfh	半精度浮点标准扩展	1.0	已批准
Zfhmin	半精度浮点最小集标准扩展	1.0	已批准
Zfinx	整数寄存器单精度浮点标准扩展	1.0	已批准
Zdinx	整数寄存器双精度浮点标准扩展	1.0	已批准
Zhinx	整数寄存器半精度浮点标准扩展	1.0	已批准
Zhinxmin	整数寄存器半精度浮点最小集标准扩展	1.0	已批准
Zmmul	整数纯乘法标准扩展	1.0	已批准
Ztso	全存储排序标准扩展	1.0	已批准

https://riscv-rtthread-programming-manual.readthedocs.io/zh-cn/latest/zh_CN/1.html#id2

supperthomas, 2025-04-18:
https://toolchains.bootlin.com/

这个平台提供toolchain都是基于linux的

supperthomas, 2025-04-18:
https://blog.csdn.net/lt6210925/article/details/147342176

supperthomas, 2025-04-18:
https://club.rt-thread.org/ask/article/6e321824382d4cc9.html