RT-Thread/rt-thread #10179: [Bug] Couldn't run board sample application

glory-man, 2025-04-10:
### RT-Thread Version

5.2.0

### Hardware Type/Architectures

GD32450Z-EVAL

### Develop Toolchain

GCC

### Describe the bug

Build & start debug sample located @ bsp\gd32\arm\gd32450z-eval 
Appl crashed (hard_fault) on 
```
rt_system_heap_init((void *)HEAP_BEGIN, (void *)HEAP_END);
```
call in `board\board.c`

[image]
next step in debugger went into hard_fault_handler
[image]

To prevent this crash I decresed 2 times available memory (in `board.h`). 
```
#define GD32_SRAM_SIZE         128
```
Ok. Program started successfully - Led1 blink. I want to see console output. Connect to COM0 (J1) via RS232-USB converter, open serial-port with putty 

[image]
but nothing happens. There is no answer from the board.

What should I do to test rtthread-sample on my board ? With `GD32_SRAM_SIZE = 256`. Or maybe need some fix on sample?

supperthomas, 2025-04-10:
do you have keil ide？

glory-man, 2025-04-11:
> do you have keil ide？

I tried to build sample with default `rtconfig.py` content. But get error

[image]

I use keil in eval mode with 32k code limitation.

So I switched to VScode + GCC + Cortex-Debug

glory-man, 2025-04-11:
> do you have keil ide？

I made only 2 modifications in `rtconfig.py`
```
CROSS_TOOL='gcc'
```
and set path to gcc-toolchain om my PC
```
EXEC_PATH   = r'C:\Users\XXYYZZ'
```

supperthomas, 2025-04-11:
you can use https://www.keil.arm.com/mdk-community/  keil community version

supperthomas, 2025-04-13:
> > you can use https://www.keil.arm.com/mdk-community/ keil community version
> 
> Why should i use keil?  I use official example distributed by rt-thread, use of gcc is provided by this example. I have not changed a single line of any supplied file. example built and ran fine on target board, only it does not work. Do you mean that if i use another IDE, then this same example will start working somehow differently?

yes maybe the author use keil to test. different ide maybe different hex bin.

glory-man, 2025-04-13:
> yes maybe the author use keil to test. different ide maybe different hex bin.

why then include the ability to use untested toolchain in the official sample?

glory-man, 2025-04-17:
I would like to get a comment from the rt-thread (or this sample) developer 
@Ouxiaolong

glory-man, 2025-04-30:
Can anybody help me ?