RT-Thread/rt-thread #10182: [Bug] GD32系列生成cmake代码

ThinkCodeStudio, 2025-04-11:
### RT-Thread Version

5.2.0

### Hardware Type/Architectures

GD32

### Develop Toolchain

Other

### Describe the bug

尝试了
* gd32303c-start
* gd32407v-lckfb

都报同一个错误
```
scons --target=cmake
scons: Reading SConscript files ...
Newlib version: 4.1.0
Update setting files for CMakeLists.txt...
AttributeError: 'CLVar' object has no attribute 'replace':
  File "E:\_rtthread\rt-thread-5.2.0\bsp\gd32\arm\gd32303c-start\SConstruct", line 60:
    DoBuilding(TARGET, objs)
  File "E:\_rtthread\rt-thread-5.2.0\tools\building.py", line 873:
    EndBuilding(target, program)
  File "E:\_rtthread\rt-thread-5.2.0\tools\building.py", line 982:
    GenTargetProject(program)
  File "E:\_rtthread\rt-thread-5.2.0\tools\building.py", line 946:
    CMakeProject(Env, Projects, GetOption('project-name'))
  File "E:\_rtthread\rt-thread-5.2.0\tools\cmake.py", line 341:
    GenerateCFiles(env, project, project_name)
  File "E:\_rtthread\rt-thread-5.2.0\tools\cmake.py", line 65:
    CFLAGS = env['CFLAGS'].replace('\\', "/").replace('\"', "\\\"")
```

ThinkCodeStudio, 2025-04-11:
windows 11 平台
ConEmu 230724

ThinkCodeStudio, 2025-04-11:
SConstruct文件中的 `CCFLAGS` -> `CFLAGS`
```python
env = Environment(tools = ['mingw'],
    AS = rtconfig.AS, ASFLAGS = rtconfig.AFLAGS,
    # CC = rtconfig.CC, CCFLAGS = rtconfig.CFLAGS,
    CC = rtconfig.CC, CFLAGS = rtconfig.CFLAGS,
    AR = rtconfig.AR, ARFLAGS = '-rc',
    CXX = rtconfig.CXX, CXXFLAGS = rtconfig.CXXFLAGS,
    LINK = rtconfig.LINK, LINKFLAGS = rtconfig.LFLAGS)
env.PrependENVPath('PATH', rtconfig.EXEC_PATH)
```

kaidegit, 2025-04-15:
bsp的问题吗，直接提一个pr呗

closed 2025-04-16 by ThinkCodeStudio