RT-Thread/rt-thread #10242: [Bug] scons --cdb生成的compile_commands.json缺少了关于rt-thread/components/libc/compilers/common/extension的包含路径

EcutAtom336, 2025-04-30:
### RT-Thread Version

master

### Hardware Type/Architectures

gd32407v-lckfb

### Develop Toolchain

Microsoft VScode

### Describe the bug

使用scons --cdb和scons --vsc后用vscode打开，clangd无法索引sys/types.h（该文件在rt-thread/components/libc/compilers/common/extension/sys/types.h），如果在生成的compile_commands.json中加上-Imy_path\rt-thread\\components\\libc\\compilers\\common\\extension就可以正常索引。gd32407v-lckfb和gd32470v-lckfb都有这个问题。
[image]

supperthomas, 2025-05-02:
stm32有这个问题吗？

EcutAtom336, 2025-05-03:
> stm32有这个问题吗？

测试了很多个stm32的bsp，无法使用scons --cdb生成compile_commands.json，报错stm32xxxx.h not found。另外测试了ev_hc32f448_lqfp80，同样无法索引sys/types.h，

supperthomas, 2025-05-03:
这个是直接用的scons接口，理论上没啥问题。可以看看为啥？

supperthomas, 2025-05-03:
https://github.com/RT-Thread/rt-thread/pull/9856

supperthomas, 2025-05-03:
stm32要pkgs --update一下

EcutAtom336, 2025-05-03:
> stm32要pkgs --update一下

[image] 还是不行

CYFS3, 2025-05-03:
> > stm32要pkgs --update一下
> 
> [image] 还是不行

换一下源在拉，menuconfig -s

EcutAtom336, 2025-05-03:
> > > stm32要pkgs --update一下
> > 
> > 
> > [image] 还是不行
> 
> 换一下源在拉，menuconfig -s

我重新克隆整个仓库，换用github源，还是一样

[image]

supperthomas, 2025-05-04:
> > > > stm32要pkgs --update一下
> > > 
> > > 
> > > [image] 还是不行
> > 
> > 换一下源在拉，menuconfig -s
> 
> 我重新克隆整个仓库，换用github源，还是一样
> 
> [image]

你这个应该是网络问题

supperthomas, 2025-05-16:
[image]

这里GCC并没有用到这里的路径，本质上应该算对的，gcc用的是toolchain的里面的头文件。

supperthomas, 2025-05-16:
extension这个文件夹也写了
```
## Attentions

This folder is "common" for toolchains, which only support ISO C, as an extension part, such as Keil-MDK and IAR.

```

loogg, 2025-05-21:
> 需要新建一个 .clangd 文件，在里面添加头文件路径，如下所示，修改为自己电脑上的路径
> 
> ```
> CompileFlags:
>   Add:
>     - -ID:/Learning/RT-Thread/env-windows/tools/gnu_gcc/arm_gcc/mingw/arm-none-eabi/include
> ```

[image]