RT-Thread/rt-thread #10248: [Bug] Unable to input Chinese.

Vincent1-python, 2025-05-07:
### RT-Thread Version

5.1.3

### Hardware Type/Architectures

HMI Board

### Develop Toolchain

RT-Thread Studio

### Describe the bug

Unable to input Chinese, it disappeared after input.

### Other additional context

Unable to input Chinese, it disappeared after input.

Vincent1-python, 2025-05-09:
Sorry, RT-Thread version is 5.0.1, and bsp version is the latest.

fengmuyou, 2025-05-16:
> Sorry, RT-Thread version is 5.0.1, and bsp version is the latest.

Could you please describe this issue in more detail? Is it specifically that RT-Thread Studio cannot input Chinese, or is it happening elsewhere?

fengmuyou, 2025-05-16:
> Sorry, RT-Thread version is 5.0.1, and bsp version is the latest.
在RT-Thread Studion 可以输入中文：
[image]

fengmuyou, 2025-05-16:
> Sorry, RT-Thread version is 5.0.1, and bsp version is the latest.
Can input Chinese in RT-Thread Studio.
[image]

Vincent1-python, 2025-05-16:
It's the serial port of HMI BOARD, so Chinese can't be input.

Rbb666, 2025-05-19:
如果是串口终端不能输入中文可以切换个串口工具试一下，比如mobaxterm

closed 2025-05-19 by Rbb666

Vincent1-python, 2025-05-19:
我就是用mobaxterm