RT-Thread/rt-thread #10268: [Bug] action CI 部分bsp编译过久

supperthomas, 2025-05-13:
### RT-Thread Version

master

### Hardware Type/Architectures

action CI

### Develop Toolchain

Other

### Describe the bug

[image]

部分bsp会卡主ci, 特别是aarch64 分组

https://productionresultssa15.blob.core.windows.net/actions-results/2763c221-bdc3-47a9-8a30-58247f6d659a/workflow-job-run-06e0143d-07a8-5dc5-8ba0-8df12160d79d/logs/job/job-logs.txt?rsct=text%2Fplain&se=2025-05-13T02%3A19%3A20Z&sig=wvqFF5R2TKsA9xozeebfhLFBYlAZhEyrT6W9ETji%2F54%3D&ske=2025-05-13T12%3A18%3A43Z&skoid=ca7593d4-ee42-46cd-af88-8b886a2f84eb&sks=b&skt=2025-05-13T00%3A18%3A43Z&sktid=398a6654-997b-47e9-b12b-9515b896b4de&skv=2025-05-05&sp=r&spr=https&sr=b&st=2025-05-13T02%3A09%3A15Z&sv=2025-05-05

[image]

supperthomas, 2025-05-13:
[    repository_url = "https://gitee.com/phytium_embedded/phytium-standalone-sdk.git"](https://github.com/RT-Thread/rt-thread/blob/3ebad06ff3055b1bdffecb1de0262b0f2dfc39a3/bsp/phytium/libraries/phytium_standalone_sdk_install.py#L5)

supperthomas, 2025-05-13:
https://github.com/RT-Thread/rt-thread/blob/3ebad06ff3055b1bdffecb1de0262b0f2dfc39a3/bsp/phytium/libraries/phytium_standalone_sdk_install.py#L5

supperthomas, 2025-05-13:
@messigogogo 你们这个仓库可否改成github的仓库？或者rtthread/package中fork一下？gitee仓库，ci下不下来。会卡住很久

supperthomas, 2025-05-13:
https://github.com/RT-Thread-packages/phytium-standalone-sdk

supperthomas, 2025-05-13:
建议参考 #9960 做成软件包的形式下载。

BernardXiong, 2025-05-13:
> https://github.com/RT-Thread-packages/phytium-standalone-sdk

需要放到`RT-Thread-packages`下吗？感觉是不是不必须的

supperthomas, 2025-05-13:
> > https://github.com/RT-Thread-packages/phytium-standalone-sdk
> 
> 需要放到`RT-Thread-packages`下吗？感觉是不是不必须的

这个厂商的SDK放在了gitee下面了，这个github CI下不下来的。那放哪里呢？

closed 2025-05-13 by supperthomas