RT-Thread/rt-thread #10455: [Feature] 自动配置can波特率的可行性

illustriousness, 2025-07-03:
### Describe problem solved by the proposed feature

目前的bsp(例如)配置CAN通信波特率时使用了大量预定义的表格 不同的芯片如果当前不支持 就要新增宏定义
是否可以考虑实现一个函数自动计算适配的波特率参数呢 例如
```c
#include <math.h>
#include <float.h>
#include <stdint.h>
#include <stdio.h>

typedef struct
{
    uint32_t prescaler;
    uint8_t  bs1;
    uint8_t  bs2;
    float    baud_error;
    float    sample_point;
} CAN_TimingConfig;

/**
 * @brief 
 * 
 * @param F_pclk CAN的APB1时钟
 * @param target_baudrate 目标波特率
 * @return CAN_TimingConfig 
 */
CAN_TimingConfig calculate_can_timing(uint32_t F_pclk,
                                      uint32_t target_baudrate)
{
    CAN_TimingConfig best                = {0};
    float            min_error           = FLT_MAX;
    float            target_sample_point = 0.8f;
    const uint8_t    tq_min = 8, tq_max = 25;
    const uint8_t    bs1_max = 16, bs2_max = 8;

    // 扩展分频范围避免遗漏
    uint32_t prescale_max = (F_pclk + target_baudrate * tq_min - 1) / (target_baudrate * tq_min);
    uint32_t prescale_min = F_pclk / (target_baudrate * tq_max);
    prescale_min          = prescale_min > 0 ? prescale_min : 1;
    prescale_max          = prescale_max > 1024 ? 1024 : prescale_max;

    // 逆序循环优先大分频值
    for (uint32_t prescale = prescale_max; prescale >= prescale_min; --prescale)
    {
        float   total_tq = (float)F_pclk / (target_baudrate * prescale);
        uint8_t tq       = (uint8_t)(total_tq + 0.5f);

        if (tq < tq_min || tq > tq_max) continue;

        uint8_t bs1 = (uint8_t)(target_sample_point * tq - 1 + 0.5f);
        uint8_t bs2 = tq - 1 - bs1;

        if (bs1 < 1 || bs1 > bs1_max || bs2 < 1 || bs2 > bs2_max) continue;

        float actual_baud  = (float)F_pclk / (prescale * (1 + bs1 + bs2));
        float baud_err     = fabsf(actual_baud - target_baudrate) / target_baudrate;
        float sample_point = (1.0f + bs1) / (1 + bs1 + bs2);
        float sample_err   = fabsf(sample_point - target_sample_point);
        float total_err    = 0.6f * baud_err + 0.4f * sample_err;

        // 误差更小 或 误差相同但分频更大 或 分频相同但TQ更小
        if ((total_err < min_error) || (total_err == min_error && ((prescale > best.prescaler) || (prescale == best.prescaler && tq < (best.bs1 + best.bs2 + 1)))))
        {
            min_error         = total_err;
            best.prescaler    = prescale;
            best.bs1          = bs1;
            best.bs2          = bs2;
            best.baud_error   = baud_err;
            best.sample_point = sample_point;
        }
    }
    return best;
}

int main()
{
    CAN_TimingConfig cfg = calculate_can_timing(36000000, 500 * 1000);
    printf("Prescaler: %u\nBS1: %u\nBS2: %u\nSample: %.1f%%\nBaud error: %.4f%%\n",
           cfg.prescaler, cfg.bs1, cfg.bs2,
           cfg.sample_point * 100, cfg.baud_error * 100);
}

```
这个函数可以考虑放在dev_can.c中 给drv_can.c传参时只需要相应的分频值和两个bs的TQ数(默认SWJ 1TQ)

illustriousness, 2025-07-03:
哦豁 刚去看了一眼bsp 发现一点小惊喜
在这个 bsp/essemi/es32f0654/drivers/drv_can.c 的  get_can_baud_index 函数有自动计算的过程
下面附上一个demo
```c
#include <stdint.h>
#include <stdio.h>

typedef struct
{
    uint32_t seg1;
    uint32_t seg2;
    uint32_t psc;
} can_init_t;

static uint32_t get_can_baud_index(uint32_t baud, can_init_t *init)
{
    /* attention !!! baud calculation example: Pclk / ((1 + seg1 + seg2) * psc)       Pclk=48 / ((1 + seg1=3 + seg2=2) * 8) = 1MHz */
    double   target, temp, min;
    uint32_t i, j, j_max, near = 0;
    target  = (double)(60000000);//Pclk=60Mhz
    target /= baud; /*计算误差1*/

    min = 0xFFFFFFFF;

    for (i = 1 + 16 + 8; i > 2; i--)     /*SYNC_SEG + SEG1 + SEG2*/
    {
        j_max = target / i / (0.98) + 1; /*缩小范围*/
        j_max = (j_max > 1024) ? (1024) : (j_max);

        for (j = target / i / 1.02; j < j_max; j++)
        {
            temp  = target / i / j; /*计算误差2*/
            temp  = (temp > 1) ? (temp - 1) : (1 - temp);
            temp += ((1.0 * i * j) / 0xFFFFFFFF);

            if (temp < min)
            {
                if (temp > 0.000001)
                {
                    near = (i << 16) + j;
                    min  = temp;
                }
                else
                {
                    init->seg1 = (uint32_t)((i - 1) * 2 / 3 - 1);
                    init->seg2 = (uint32_t)(i - init->seg1 - 1 - 1 - 1);
                    init->psc  = j;

                    return 0;
                }
            }
        }
    }

    if (min < 0.01)
    {
        i          = near >> 16;
        j          = near % (1 << 16);
        init->seg1 = (uint32_t)((i - 1) * 2 / 3 - 1);
        init->seg2 = (uint32_t)(i - init->seg1 - 1 - 1 - 1);
        init->psc  = j;

        return 0;
    }
    else
    {
        return 1;
    }
}

int main()
{
    can_init_t cfg = {0};

    int res = get_can_baud_index(1000 * 1000, &cfg);

    if (res == 0)
    {
        printf("Prescaler: %u\nBS1: %u\nBS2: %u\n", cfg.psc, cfg.seg1, cfg.seg2);
    }
}

```
刚简单验证了一下1M和500K 一切正常 ^_^

a1012112796, 2025-07-07:
可以但意义不大，can 通信一般常用的就那么几个波特率，搞一套如此复杂的逻辑有点多余。不过具体看bsp 如果实现吧。

wdfk-prog, 2025-07-08:
那做成软件包提供咯^-^