RT-Thread/rt-thread #10497: [Bug] scons 用armclang编译问题

liqingxin0092, 2025-07-11:
### RT-Thread Version

v5.2.1

### Hardware Type/Architectures

stm32的 bule pill，星火一号

### Develop Toolchain

Other

### Describe the bug

我想让scons编译代码，用armclang的工具链，设置好keil和keil路径， bsp目录下的rtconfig.py也改成了arm clang，遇到的问题就是代码里的.s文件也用编译c的工具编译了.
[image]

[image]
rtconfig.py里没改过关于编译s用的什么工具,

[image]

[image]

supperthomas, 2025-07-15:
你最后如果想用keil的arm clang，命令还是'scons --target=mdk5   设置好keil路径即可。

supperthomas, 2025-07-16:
https://github.com/RT-Thread/rt-thread/blob/master/.github%2Fworkflows%2Fbsp_buildings.yml#L133-L133

closed 2025-07-16 by liqingxin0092