RT-Thread/rt-thread #10501: [Bug] rt-smart(riscv64)信号处理函数出现load page fault以及pthread_cancel无法取消线程问题

eatvector, 2025-07-13:
### RT-Thread Version

master

### Hardware Type/Architectures

qemu-virt64-riscv

### Develop Toolchain

GCC

### Describe the bug

1. 运行下面测试代码，linux可以正确取消终止线程而rt-smart下无法取消终止线程，还可能出现load page fault,由于rt-smart 的riscv64工具链不支持取消点检测（见#10492），手动插入 __testcancel()：

“”“
#include <stdio.h>
#include <stdlib.h>
#include <stdint.h>
#include <pthread.h>
#include <unistd.h>
#include <errno.h>
#include <time.h>

pthread_t tid;

static void *subthread(void *arg)
{
    while (1) {
        // 取消点检测
        __testcancel();
        printf("%s: %d\n", __func__, __LINE__);
        usleep(1000000);
    }

    return 0;
}

int main(int argc, char **argv)
{
    pthread_create(&tid, NULL, subthread, 0);

    usleep(100);

    printf("%s: %d\n", __func__, __LINE__);
    pthread_cancel(tid);
    printf("%s: %d\n", __func__, __LINE__);
    pthread_join(tid, NULL);
    printf("%s: %d\n", __func__, __LINE__);

    return 0;
}

”“”

2 问题定位

出现load page fault：定位到是由于信号处理函数会访问线程TLS数据，risc-v架构下通过tp寄存器寻址，而
内核将控制流定位到用户态信号处理函数时没有恢复线程tp指针，已提交解决pr：

pthread_cancel无法取消线程：定位到是子线程线程一直反复执行cancel_handler信号处理函数，而无法继续执行第一次被信号中断的后续代码，导致其无法运行到延迟取消点，从而无法终止，已提交解决pr：

unicornx, 2025-07-23:
请问 #10500 是否解决了这个 issue？这个 issue 可以关闭了吗？

eatvector, 2025-07-23:
> 请问 [#10500](https://github.com/RT-Thread/rt-thread/pull/10500) 是否解决了这个 issue？这个 issue 可以关闭了吗？

是的已经解决，可以关闭了

unicornx, 2025-07-23:
> 是的已经解决，可以关闭了

你自己关吧，我只是路过问一下 :)

eatvector, 2025-07-23:
> > 是的已经解决，可以关闭了
> 
> 你自己关吧，我只是路过问一下 :)

好的

closed 2025-07-23 by eatvector