RT-Thread/rt-thread #10513: [Feature] RFC：整理内核单元测试

unicornx, 2025-07-16:
### Describe problem solved by the proposed feature

目前内核的单元测试程序代码有很大一部分都集中放在 `examples/utest/testcases`, 首先直观上看上去不合适，单元测试应该提高地位，而不是作为 example 存在，其次目前单元测试并没有 maintainer 认领和维护，这看上去是个问题。

我建议将测试用例整理后分类，以后都放到各自归属的模块下面去，譬如 `src/klibc/utest` 这个是当时满老师整理的一个例子，当时记得开会讨论时他也建议以后 unit test case 按照各自模块自己负责的方式存放。

但考虑到单元测试有统一管理的需要（主要是需要配置测试开关），我觉得所谓的统一维护可以体现在 Kconfig 上，即可以将内核的所有模块的单元测试的配置维护一个统一的入口菜单项，但是具体的测试代码还是放在各个模块里自己维护，这样以后每个模块如果有 maintainer 认领后，这个 maintainer 自然也就担负起对本模块的单元测试的维护工作。

unicornx, 2025-08-26:
#10534 为实现该方案搭建了一个初始框架，以后的 utest 的 testcase 源码会分别放到各个模块下自己维护，但是 Kconfig 的入口会放在 `Kconfig.utestcases` 中统一维护。这样在执行诸如 menuconfig 时可以统一从一个地方进入进行配置，避免在 menuconfig 界面中到处寻找 utest 的配置开关。

对于各个模块来说，以后可以按照如下方式统一维护单元测试用例：
- 在自己的模块所在目录下新建一个名为 `utest` 的子目录
- 在 utest 子目录中存放以下文件：
   - 本模块的单元测试用例程序代码 c 文件。
   - Kconfig 文件，为本模块的单元测试文件增加配置选项，推荐选项命名为 `RT_UTEST_TC_USING_XXXX`，XXXX 为本模块的全局唯一模块名。
   - SConscript 文件，注意在增加 src 文件时，除了依赖于 `RT_UTEST_TC_USING_XXXX` 还要依赖 `RT_UTEST_USING_ALL_CASES`，两者是 or 的关系。`RT_UTEST_USING_ALL_CASES` 的作用是一旦打开这个选项，则所有的单元测试都会启用，避免逐个选择。

以上工作做好后，可以将本模块的 Kconfig 文件的路径通过 rsource 的方式添加到 `Kconfig.utestcases` 中，这样我们在 menuconfig 时就会看到一个统一的和 utestcases 相关的入口菜单。

需要注明的是: 等以后我们把 example 目录下所有的测试用例代码都移到其他目录下后，这时候才可以彻底地删除 `bsp_update_kconfig_testcases` 和 `exclude_utestcases` 这个函数以及调用他们的地方。

kurisaW, 2025-09-02:
TODO：文档中心同步更新