RT-Thread/rt-thread #10528: [Bug] 使用 rt_object_find()时，当 name 的长度 > RT_NAME_MAX, 查找不到，同之前的行为不一致

dongly, 2025-07-20:
### RT-Thread Version

ece19e961aaa5484e1e48a398b9bc4a2e54f9ed9

### Hardware Type/Architectures

stm32f407

### Develop Toolchain

GCC

### Describe the bug

这是 #10523  引入的，不兼容旧版本行为。
如：
在默认情况下，RT_NAME_MAX == 8,
`rt_sfud_flash_find_by_dev_name("norflash0")` ,原来是可以查找到的，现在就查找不到了

kurisaW, 2025-07-21:
这个更建议修改RT_NAME_MAX大小吧

CYFS3, 2025-07-21:
[image]
这边使用的是rt_strncmp,应该使用RT_NAME_MAX来比较的，"norflash0"，比较"norflas"，会不一样，后者第七位是\0,前者则是‘h’。
看要怎么做统一，因为其实就算没有存\0，但是系统都是用rt_strncmp，来区分也不会用溢出的问题吧。

dongly, 2025-07-21:
我初步在这里改为 rt_strncmp(obj->name, name, RT_NAME_MAX -1) `, 通过了，未充分验证

kurisaW, 2025-07-24:
@dongly 麻烦帮忙测试一下，另外附有扩展了object的测试用例（rt_thread_find、rt_device_find、rt_object_find）

* [object.txt](https://github.com/user-attachments/files/21398136/object.txt)
* [object_tc.txt](https://github.com/user-attachments/files/21398143/object_tc.txt)

closed 2025-07-25 by kurisaW