RT-Thread/rt-thread #10594: [Bug] 选择 使用 RT_USING_SERIAL_V2时 , 同时使能DMA,编译出现多个宏定义缺失

GPTKEY, 2025-08-09:
### RT-Thread Version

最新版

### Hardware Type/Architectures

stm32/bluepill

### Develop Toolchain

MDK Keil

### Describe the bug

选择了  RT_USING_SERIAL_V2
 使能了 Enable serial DMA mode

compiling drv_usart_v2.c...
libraries\HAL_Drivers\drivers\drv_usart_v2.c(933): error:  #20: identifier "BSP_UART1_RX_BUFSIZE" is undefined
      uart_obj[UART1_INDEX].serial.config.rx_bufsz = BSP_UART1_RX_BUFSIZE;
libraries\HAL_Drivers\drivers\drv_usart_v2.c(934): error:  #20: identifier "BSP_UART1_TX_BUFSIZE" is undefined
      uart_obj[UART1_INDEX].serial.config.tx_bufsz = BSP_UART1_TX_BUFSIZE;
libraries\HAL_Drivers\drivers\drv_usart_v2.c(937): error:  #20: identifier "BSP_UART1_DMA_PING_BUFSIZE" is undefined
      uart_obj[UART1_INDEX].serial.config.dma_ping_bufsz = BSP_UART1_DMA_PING_BUFSIZE;
libraries\HAL_Drivers\drivers\drv_usart_v2.c(939): error:  #20: identifier "UART1_RX_DMA_INSTANCE" is undefined
      static struct dma_config uart1_dma_rx = UART1_DMA_RX_CONFIG;
libraries\HAL_Drivers\drivers\drv_usart_v2.c(939): error:  #20: identifier "UART1_RX_DMA_RCC" is undefined
      static struct dma_config uart1_dma_rx = UART1_DMA_RX_CONFIG;
libraries\HAL_Drivers\drivers\drv_usart_v2.c(939): error:  #20: identifier "UART1_RX_DMA_IRQ" is undefined
      static struct dma_config uart1_dma_rx = UART1_DMA_RX_CONFIG;
libraries\HAL_Drivers\drivers\drv_usart_v2.c: 0 warnings, 6 errors

Rbb666, 2025-08-11:
欢迎PR修复下

illustriousness, 2025-08-11:
看起来需要你在board.h或者rtconfig 把这些缺少的宏定义写一下

closed 2025-08-25 by kurisaW