RT-Thread/rt-thread #10649: [Feature] [STM32][Kconfig] startup_xx.s的设置

wdfk-prog, 2025-09-01:
### Describe problem solved by the proposed feature

- 例如startup_stm32f407xx.s的文件是通过cubemx生成(大概率).然后链接到RTT工程中的.
- 这样每次生成CUBEmx代码都会被修改
- 应该在源码中提供修改后的startup_xx.s启动文件.而不是要手动修改或者去模板工程中寻找

- 根据现在STM32hal库做成软件包的形式的话,可以在SConscript添加一个.s的链接
> https://github.com/RT-Thread-packages/stm32f4_hal_driver/blob/master/SConscript

例如:
```
if GetDepend(['SOC_STM32F407']):
    src += ['CMSIS/Device/ST/STM32F4xx/Source/Templates/gcc/startup_stm32f407xx.s']
elif GetDepend(['SOC_STM32F405']):
    src += ['CMSIS/Device/ST/STM32F4xx/Source/Templates/gcc/startup_stm32f405xx.s']
```

- 因为默认的HAL库有提供一份.s启动文件,修改这个文件既可,也不需要进行修改了.

wdfk-prog, 2025-09-02:
> 一份完整的STM32驱动库包含两部分：CMSIS+HAL-Drivers
> 
> * https://github.com/RT-Thread-packages/hc32-f4-cmsis/blob/master/SConscript
> * https://github.com/RT-Thread-packages/stm32f4_hal_driver/blob/master/SConscript
> 
> 你说的这个应该有的吧，另外导入到cubemx中重新生成后，官方的cmsis和hal-driver都是不使用的，默认建议用户主动删除，最关键只是需要一个stm32fxxx_hal_msp.c而已

- .s并无在脚本中链接,而是使用cubemx生成的.s.
- 问题就是cubemx每次重新生成,那么.s也会被改动

kurisaW, 2025-09-02:
那重点问题应该是链接的问题，之前我看也有人反馈过链接不上，是windows环境还是linux？

kurisaW, 2025-09-02:
是的原本的想法就是这样，不过最近我看主线的tools这块有一些PR改动，可能导致语法识别失败了

wdfk-prog, 2025-09-02:
> 是的原本的做法就是这样，不过最近我看主线的tools这块有一些PR改动，可能导致语法识别失败了

> https://github.com/RT-Thread-packages/stm32f4_cmsis_driver/blob/master/SConscript

现在应该就是用cmsis里面的.s了

kurisaW, 2025-09-02:
所以使用上还有什么疑问吗，没有的话可以关闭此issue了

wdfk-prog, 2025-09-02:
> 所以使用上还有什么疑问吗，没有的话可以关闭此issue了

十分Thanks♪(･ω･)ﾉ