RT-Thread/rt-thread #10680: [Feature] Utest 测试用例推进事项

Rbb666, 2025-09-10:
### Describe problem solved by the proposed feature

由于目前RTT关于组件相关的 UTEST 测试用例较为稀少，为了更好的针对组件进行维护，需要针对重点组件进行测试用例的补充，目前准备推进的测试用例包括以下内容：

期望达成的目标：
 
* 覆盖到90%以上的API
* 部署到CI（qemu）上运行测试

## 文件系统

- [x] DFSV1/V2相关API：https://github.com/RT-Thread/rt-thread/pull/10670
- [x] POSIX文件系统API：https://github.com/RT-Thread/rt-thread/pull/10670

## 网络

- [ ] LWIP 相关API：https://github.com/RT-Thread/rt-thread/pull/10675
- [ ] NETDEV相关API
- [ ] SAL 相关API
- [ ] AT组件

## C++

- [ ] ipc
- [ ] thread
- [ ] cpp11

## utilities

- [ ] ulog

Ryan-CW-Code, 2025-09-10:
支持，但感觉不要着急，先完善一下测试框架。
比如引入模糊测试，替代目前手动实现或者没有实现的随机数据生成方式。
申请Coverity Scan对RTT进行检查，开源项目免费，还是能发现一些cppcheck查不到的错误。
在qemu测试时把Sanitizer打开，在muc跑时使用RTT的内存检查，没有的话可以提供一下
还有引入coderabbitai审查，它会比copilot更详细一些，很多前端项目都在使用，对C的支持也很好

可以参考zephyr的实践，与utest结合一下，希望RTT越来越好😊

Ryan-CW-Code, 2025-09-10:
Coverity Scan 据我了解没有很方便和CI进行集成的方法，只能维护者手动进行操作。
zephyr官方默认SCA是Coverity Scan，两周手动执行一次，然后用脚本添加到issue。

linux和freertos的静态分析也在使用 Coverity Scan ， 我体验过是比cppcheck和clang-tidy强的，就是稍微有点麻烦，需要先申请，还要下载软件，生成文件再上传，然后等待分析结果，还需要手动或者脚本添加到issue中

closed 2025-10-23 by Rbb666