RT-Thread/rt-thread #10681: [讨论] 如何确保单元测试用例名字的唯一性

unicornx, 2025-09-10:
### Describe problem solved by the proposed feature

。。。

unicornx, 2025-09-10:
@Rbb666 @kurisaW 欢迎 comment

Rbb666, 2025-09-10:
个人感觉：`测试用例目录.测试用例名称` 这样命名会清晰一点

比如：

* components/dfs/utest/tc_dfs_api.c 的写法：`components.dfs.tc_dfs_api`
* src/klibc/utest/tc_rt_memcmp.c 的写法：`src.klibc.tc_rt_memcmp`

unicornx, 2025-09-10:
> 个人感觉：`测试用例目录.测试用例名称` 这样命名会清晰一点

我感觉你说的这个和我提的其实差不多啊，区别仅仅在于
- 你说的 “测试用例目录” 其实就是我说的 “模块前缀”，我提的 “模块前缀” 的格式就是基于路径名，把 “/” 换成 “.”。和你的想法是一回事。具体可以看我的举例子。而我对 src 这个希望作为一个特列，因为我觉得 src 这个目录名字不好看而已，当然如果不做特别要求，src 我也没意见。（说实在的我觉得原来代码目录里这个 src 真心是个不好的名字）
- 你说的“测试用例名称” 从你举的例子来看似乎和文件名绑定了，而我的建议的第二部分 “测试功能” 其实对于这个要求更松一些，只要确保在同一个 “模块前缀” 范围内唯一即可，从现有的代码来看，大家其实都没有实现文件名和测试用例名一致，我觉得没有必要强制要求一致，否则改起来改动太大。