RT-Thread/rt-thread #10716: [Bug] STM32的SPI驱动在DMA 全双工模式下 recv_buf 数据未更新的问题

zx115, 2025-09-20:
### RT-Thread Version

5.1.0

### Hardware Type/Architectures

STM32F407VGT6

### Develop Toolchain

MDK Keil

### Describe the bug

#### 现象复现：

1. 在menuconfig界面下使能`BSP_SPIx_TX_USING_DMA`及`BSP_SPIx_RX_USING_DMA`；
  
2. 使用`rt_malloc()`函数申请两块内存：tx_data(>=10)和rx_data(>=10)；
  
3. 将目标SPI总线的MISO接口拉高或拉低；
  
4. 调用`rt_spi_transfer`函数向SPI设备发送>=10字节的数据，使用上一步申请的内存tx_data和rx_data分别作为发送缓存和接收缓存；
  
5. 打印rx_data中的内容，发现内容不符合预期。期望结果应为rx_data的值全为0xFF（MISO接口拉高）或0x00（MISO接口拉低），而实际则为内存中未更新的随机数据。
  

#### 问题定位：

问题出在 `rt-thread-5.1.0/bsp/stm32/libraries/HAL_Drivers/drivers/drv_spi.c`的`spixfer`函数中。

#### 原因分析：

- 348行到374行的分析：
  

在启用TX_DMA的情况下，为了提高DMA的效率，该函数对`send_buf`的值（即发送缓冲区的地址）做了对齐4字节的操作，当`send_buf`的值恰巧对齐4字节时，直接将send_buf的值赋给`p_txrx_buffer`；当不对齐时，则通过`rt_malloc()`函数重新申请一块内存(该函数能够保证生成的内存对齐4字节)，并将该地址赋给`p_txrx_buffe`，同时也保存一份副本在`dma_aligned_buffer`变量里(在此之前该变量的值为`RT_NULL`)。

- 383行的分析：
  

在TX_DMA和RX_DMA皆启用的情况下，数据的传输实际上是调用了`HAL_SPI_TransmitReceive_DMA`函数实现的。但是在这个驱动中使用`p_txrx_buffe`同时作为`HAL_SPI_TransmitReceive_DMA`的TX和RX缓存，也就是说`HAL_SPI_TransmitReceive_DMA`在发送`p_txrx_buffe`中的数据的同时也在将接收到的数据写入其中，这本身并没有太大问题，只要后面再将接收到的数据复制回`recv_buf`中就行。

- 468行到482行的分析：
  

但是这个函数并不是在任何情况下都会将接收到的数据复制回`recv_buf`中，而是只在`dma_aligned_buffer != RT_NULL`的情况下才会执行`rt_memcpy(recv_buf, p_txrx_buffer, send_length);`语句将数据复制回去。这里的`dma_aligned_buffer`就是前面提到的重新申请的内存的副本。只有当`send_buf`的值不对齐4字节时，`dma_aligned_buffer`才会保存一份重新申请的内存的副本，而如果`send_buf`的值恰巧对齐4字节，`dma_aligned_buffer`将维持初始值`RT_NULL`，此时`rt_memcpy(recv_buf, p_txrx_buffer, send_length);`语句也将无法得到执行，于是就造成了`recv_buf`不更新的bug。

#### 其他问题：

事实上，这份驱动还有其他问题。例如，如果你只使能了TX_DMA而未使能RX_DMA，则`spixfer`则只使用语句`HAL_SPI_Transmit_DMA(spi_handle, (uint8_t *)p_txrx_buffer, send_length);`将数据发送出去，而完全抛弃了接收数据。在这种情况下，如果上层应用调用框架接口`rt_spi_recv`单纯只接收数据，将出现接收数据错误的问题。

wdfk-prog, 2025-09-21:
- 使用spi的最新代码试一下,5.1.0之后还有有点修改的.https://github.com/RT-Thread/rt-thread/pull/8836

zx115, 2025-09-21:
> * 使用spi的最新代码试一下,5.1.0之后还有有点修改的.[fix:[DMA SPI]特定条件下接收错误 #8836](https://github.com/RT-Thread/rt-thread/pull/8836)

我试了一下5.2.1版的drv_spi，依然有问题，问题的原因还是我上面分析的：当`message->send_buf`的值已经4字节对齐时，拷贝功能的代码`rt_memcpy(recv_buf, p_txrx_buffer, send_length);`就不会得到执行。

wdfk-prog, 2025-09-21:
> > * 使用spi的最新代码试一下,5.1.0之后还有有点修改的.[fix:[DMA SPI]特定条件下接收错误 #8836](https://github.com/RT-Thread/rt-thread/pull/8836)
> 
> 我试了一下5.2.1版的drv_spi，问题依然存在，问题的原因还是我上面分析的：当`message->send_buf`的值已经4字节对齐时，拷贝功能的代码`rt_memcpy(recv_buf, p_txrx_buffer, send_length);`就不会得到执行。

确实有问题....

wdfk-prog, 2025-09-22:
> https://github.com/RT-Thread/rt-thread/pull/9124

我做了修改,可以试一下是否修复了bug