RT-Thread/rt-thread #10722: [Feature] RISC-V TODO List

MuChenger, 2025-09-22:
### Describe problem solved by the proposed feature

**内核**
1. 优化 libcpu/risc-v/common 目录下的文件，将当前浮点寄存器无差别压栈的方式改为根据浮点寄存器的使用情况选择性压栈。待修改文件路径如下：
https://github.com/RT-Thread/rt-thread/tree/master/libcpu/risc-v/common
2. 优化 libcpu/risc-v/common 目录下的文件，添加 rvv0.7/1.0 支持。
3. RISC-V RVA23 Profile 兼容性支持

**BSP**
1. qemu-virt64-riscv BSP 添加对 smp 的支持，bsp 的目录如下：
https://github.com/RT-Thread/rt-thread/tree/master/bsp/qemu-virt64-riscv
2. SpacemiT K1 适配 RT-Thread BSP 并可支持 SMP 模式。
3. 验证RISC-V 相关 BSP 是否可以以导入 BSP 的方式在 RT-Studio 上的编译，并修复出现的问题。

**组件**
1. 移植面向 RISC-V 的调试，性能分析等类型的组件

BernardXiong, 2025-11-16:
好像 qemu-virt-riscv 是支持 rv32、rv64的，或许可以让 qemu-virt-riscv 更好的使用起来。另外，关于如何有一份尽可能小，基本，资源占用低的rt-thread，也可以多多考虑到 <可以通过什么样的方式？也包括使用mlibc？>

PiyushPatle26, 2026-09-04:
Hi, I recently sent an email proposing an OSPP project around RISC-V lazy FP/RVV context switching. I noticed that this issue already lists related TODOs for selective FP saving and RVV support.

Would it be possible to consider a focused subset of these items as an OSPP 2026 project, initially targeting single-hart RISC-V?
I’d be happy to prepare a detailed proposal if this direction is suitable.

closed 2026-10-17 by Rbb666